smartdevicelink/sdl_core #1601: UpdateTurnList: SDL sends "turnText" instead of "navigationText"

DrachenkoAnastasiia, 2017-06-07:
## Occurrence
Always

## Preconditions
1. SDL is built with Proprietary flag.
2. SDL and HMI are started. First ignition cycle.
3. Connect device
4. Register new application.

## Steps to reproduce
1. App->SDL: UpdateTurnList()   
```
turnList = {
                  {
                    navigationText ="Text",
                    turnIcon =
                            {
                            value ="icon.png",
                            imageType ="DYNAMIC",
                            },
                  }
              }
```
SDL->HMI:Navigation.UpdateTurnList()

## Actual result
```
navigationText = {fieldName = "turnText", fieldText = "Text"}
```

## Expected result
```
navigationText = {fieldName = "navigationText", fieldText = "Text"}
```

# Environment
SDL: https://github.com/LuxoftAKutsan/sdl_core/tree/integration/4.3.1 (35344c8)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (fe002fb)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (4ba374c)
OS: Ubuntu 14.04 LTS x64

# Attachments
[SDLOPEN-2470.zip](https://github.com/smartdevicelink/sdl_core/files/1058182/SDLOPEN-2470.zip)

# Expected delivery   
- [ ] Source code updates 
- [ ] Code comments 
- [ ] UTs add/update 
- [ ] ATF tests add/update ***(not required)***  
- [x] Manual tests ***(not required)*** 
- [x] Add/update CI plans/jobs ***(not required)***  
- [x] SDD updates ***(not required)***  
- [x] Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) ***(not required)***  
- [x] Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines))  ***(not required)***

iCollin, 2021-03-31:
Closed as `navigationText` is not a `TextFieldName`, but the name of the HMI API String parameter

closed 2021-03-31 by iCollin

VNikiforov-dev, 2021-04-21:
The sequence from the issue is covered in [027_UpdateTurnList_PositiveCase_SUCCESS](https://github.com/LuxoftSDL/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/API/027_UpdateTurnList_PositiveCase_SUCCESS.lua)