smartdevicelink/sdl_core #1608: The OnEventChanged notification provides the wrong state info during app activation

mrapitis, 2017-06-13:
**Description:**
When an app is activated, the onEventChanged notification erroneously does not distinguish between different App types (Media, Non-Media, Navigation) -- this eventually leads to wrong onHMIStatus notifications sent to the mobile app.

**Occurrence:**
Always

**Use Cases:**
Example use cases where sdl core is processing the OnEventChanged notification incorrectly:

- A navigation app type is activated during an active embedded audio source or embedded navigation session
- A media app is activated during an active embedded audio source or embedded navigation session
- Navigation and media app perform audio streaming during active embedded audio and navigation source selection
- A non-media app is activated during active embedded audio source or navigation selection
- Multiple app activated during active embedded navigation or audio source selection

**Reproduction Steps and Observed Behavior:**
Ensure a navigation app type is running and has a FULL and AUDIBLE state
Activate an embedded audio source
Take note of the following messaging

- HMI -> SDL: OnEventChanged (AUDIO_SOURCE, isActive=true)
- SDL -> NAVI app: OnHMIStatus (LIMITED, **NOT_AUDIBLE**)

Activate a navigation type app
Take note of the following messaging
- HMI -> SDL: SDL.ActivateApp (<appID>)
- SDL -> HMI: SDL.ActivateApp (SUCCESS)
- **HMI -> SDL: OnEventChanged (AUDIO_SOURCE, isActive=false)**
- SDL -> NAVI app:OnHMIStatus (FULL, AUDIBLE)

**Expected Behavior:**
Ensure a navigation app type is running and has FULL and AUDIBLE state
Activate an embedded audio source
**Please take note of the changes in messaging below:**

- HMI -> SDL: OnEventChanged (AUDIO_SOURCE, isActive=true)
- SDL -> NAVI app: OnHMIStatus (LIMITED, **AUDIBLE**)

Activate a navigation type app
Take note of the following messaging
- HMI -> SDL: SDL.ActivateApp (<appID>)
- SDL -> HMI: SDL.ActivateApp (SUCCESS)
- SDL -> NAVI app:OnHMIStatus (FULL, AUDIBLE)

theresalech, 2017-06-28:
In June 27 SDLC Steering Committee Meeting, Steering Committee agreed that this should be high priority.

dboltovskyi, 2020-06-15:
Issue is very similar to https://github.com/smartdevicelink/sdl_core/issues/1906

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

closed 2020-09-04 by jacobkeeler