smartdevicelink/sdl_core #1654: SDL Core, could not be installed, No appender could be found for logger (SDLMain). log4cxx: Please initialize the log4cxx system properly.

varunparashar921, 2017-07-09:
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "SmartDeviceLinkCoreLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "ProtocolFordHandlingLogFile".
log4cxx: Could not instantiate class [SafeFileAppender].
log4cxx: Class not found: SafeFileAppender
log4cxx: Could not instantiate appender named "TransportManagerLogFile".
log4cxx: No appender could be found for logger (SDLMain).
log4cxx: Please initialize the log4cxx system properly.

Bigtail12002, 2017-11-01:
I also encountered this issue. 
https://github.com/smartdevicelink/sdl_core/issues/809
I tried to run under build/bin/ , but it still failed.

jacobkeeler, 2018-03-08:
@varunparashar921 Closing for now due to a lack of information. The latest releases provide a start script to simplify this (using this, you don't need to manually include `LD_LIBRARY_PATH=.`) [here](https://github.com/smartdevicelink/sdl_core/blob/master/src/appMain/start.sh).

Feel free to reopen if you are still having this issue.

closed 2018-03-08 by jacobkeeler