smartdevicelink/sdl_core #1671: [SDL 0073] Adding Metadata Types

theresalech, 2017-07-19:
Proposal: [Adding Metadata Types](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0073-Adding-Metadata-Types.md)

> This proposal introduces metadata types to the text field struct in order to allow for more robust HMI designs.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/208

Steering Committee Decision:

> Now that the requested revisions have been incorporated, the Steering Committee has agreed to accept this proposal.

theresalech, 2017-08-16:
Updated with [evolution pull request.](https://github.com/smartdevicelink/sdl_evolution/pull/239).

Modify the names in the proposal to the following:

Enum: From TextFieldType to MetadataType
Struct: From MetadataStruct to MetadataTags
Show Param: From textFieldMetadata to metadataTags:

theresalech, 2017-08-31:
Updated with two Evolution Pull Requests:

1. [Remove "none" from Metadata enumeration](https://github.com/smartdevicelink/sdl_evolution/pull/259).

Motivation for Changes:
> The "none" tag is redundant as it is equivalent to sending an empty TextFieldType/MetadataType array. Also, the "none" was inconsistently only defined in the HMI_API but never for the MOBILE_API.

Summary of Changes:
> The proposed solution is to remove "none" element.

Full details can be found in the [evolution pull request](https://github.com/smartdevicelink/sdl_evolution/pull/259).

2. [Change TextFieldType to MetadataType in HMI_API]
(https://github.com/smartdevicelink/sdl_evolution/pull/266)

Motivation for Changes:
>With the change from `TextFieldType` to `MetadataType` in the Mobile API, the HMI API should be updated accordingly

Summary of Changes:
> The proposed solution is to change the name of the `TextFieldType` enum to `MetadataType` in the HMI API.

Full details can be found in the [evolution pull request](https://github.com/smartdevicelink/sdl_evolution/pull/266)

jacobkeeler, 2017-09-07:
Closing with the merge of #1746

closed 2017-09-07 by jacobkeeler