smartdevicelink/sdl_core #1683: [SDL 0080] Support for MultiSession protocol string

theresalech, 2017-07-26:
Proposal: [Support for MultiSession protocol string](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0080-Support-for-MultiSession-protocol-string.md)

> This proposal introduces a new protocol string for head units which support iOS multisession (Ability for multiple apps to connect over the same protocol string) and also identifies the changes to the iOS proxy layer.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/228

theresalech, 2017-11-08:
@robinmk this has been implemented in iOS 4.7 (https://github.com/smartdevicelink/sdl_ios/issues/662), but we are unclear on the necessary chagnes for Core.  Can you please advise?

robinmk, 2017-11-08:
The changes on the system side should be on the transport adapter layer and for publishing of this string when the iAP session is setup with the device.

jacobkeeler, 2018-03-01:
@robinmk I assume you are referring to the iAP transport adapter, which isn't present in the open version of SDL Core. I don't see any references to any protocol strings at all in this project apart from an unused INI option. It appears that no changes will be necessary in SDL Core for this proposal.

AStasiuk, 2018-03-07:
Agreed with @mrapitis, @jacobkeeler.

closed 2018-03-19 by jacobkeeler