smartdevicelink/sdl_core #1685: [SDL 0075] OEM specific Human Interface Device support as Plug-in architecture in SDL proxy

theresalech, 2017-07-26:
Proposal: [OEM specific Human Interface Device support as Plug-in architecture in SDL proxy](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0075-HID-Support-Plug-in.md)

> Many OEM head units (ex. Lexus Remote Touch, Mazda Commander Control, BMW Gesture Control, Audi MMI) do not support direct touch interaction, but rather support a focus and select model driven by physical controls. This proposal describes a proxy plug-in interface that models the physical controls as HID devices with OEM-specific plug-in implementations. Note: From conclusion in Steering Committee, this proposal was changed to alt#1 (specify proper RPC, not plug-in architecture).

Review: https://github.com/smartdevicelink/sdl_evolution/issues/219

Steering Committee Decision:

> The Steering Committee has agreed to move forward with a revised version of Alternative 1 for this proposal, and focus on RPC changes only.  While this would typically warrant "returning for revisions," the Steering Committee has decided to make an exception and "accept with revisions" so that the feature can meet the deadline to be contained within the Core 4.4 Release.

This proposal has been accepted with the following revisions, to be made by the Xevo team:

- Alternative 1 will be the planned approach, with focus only on RPC changes - a separate proposal describing a fuller structure around proxy implementation should be entered at a later date
- Add ID to `SpatialStruct`
- Change `HapticSpatialData` array from mandatory to optional
- Increase the maximum size of the `HapticSpatialData` array (to at least 100, possibly 999)
- Add the following to the description of `SendHapticData` RPC: when a request is sent, if successful, it will replace all spatial data previously sent through RPC

These changes have now been reflected in .md file.

joeygrover, 2017-08-09:
Updated with [evolution pull request.](https://github.com/smartdevicelink/sdl_evolution/pull/234).
Add a single param to the `VideoStreamingCapability` struct:

```xml
    <struct name="VideoStreamingCapability">
        <description>Contains information about this system's video streaming capabilities.</description>
        ....
        <param name="hapticSpatialDataSupported" type="Boolean" mandatory="false">
            <description>True if the system can utilize the haptic spatial data from the source being streamed. </description>
        </param>
    
    </struct>

```

theresalech, 2017-08-23:
Updated with two evolution pull requests, accepted on August 22.

1. [Change `HapticSpatialData` parameter name to `hapticSpatialData`](https://github.com/smartdevicelink/sdl_evolution/pull/253)

>Summary of changes:
Change the name of the HapticSpatialData parameter to hapticSpatialData

2. [Change name of `SpatialStruct` to `HapticRect`](https://github.com/smartdevicelink/sdl_evolution/pull/254)

>Summary of changes:
- Change `SpatialStruct` to `HapticRect`
- Change `SendHapticData` parameter name to `HapticRectData`
- Remove the xy width and height from `HapticRect` struct into its own struct called `Rectangle`
- Change descriptions to better reflect `Rectangle`

jacobkeeler, 2017-09-05:
Resolved with the merge of #1712

closed 2017-09-05 by jacobkeeler