smartdevicelink/sdl_core #1693: SDL core dump if send End Service control frame to one of the 2 sessions registered

dsmyrnov, 2017-07-28:
Environment:
OS: Ubuntu 16.04.
ARC: PC (x86)
SDL - Tag 4.2.1 commit e98cef9
Occurrence:
Always
Preconditions:
SDL is running and HMI is connected to SDL
Steps to reproduce:
Register two application with SDL
Send End Service Control Frame from one of the sessions
Actual result:
SDL core dump after first end service control frame
Expected result:
SDL sends End Service Ack to this session
There is no dump
[core.zip](https://github.com/smartdevicelink/sdl_core/files/1182771/core.zip)
[SmartDeviceLinkCore.txt](https://github.com/smartdevicelink/sdl_core/files/1182773/SmartDeviceLinkCore.txt)

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Can the reproduction steps described actually reproduce this issue? Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@dsmyrnov, can you please advise if you are seeing this issue when using Core 6.1?

dboltovskyi, 2020-05-11:
@theresalech Please notice this issue is not actual for Core 6.1.

theresalech, 2020-05-11:
Closing as not reproducible in latest version of Core.

closed 2020-05-11 by theresalech