smartdevicelink/sdl_core #1694: SDL respond with not supported resultCode = UNSUPPORTED_RESOURCE for ChangeRegistration

dsmyrnov, 2017-07-28:
Environment:
OS: Ubuntu 16.04.
ARC: PC (x86)
SDL - Tag 4.2.1 commit e98cef9
Occurrence:
Always
Preconditions:
SDL is running and HMI is connected to SDL
Application registered
Steps to reproduce:
Send ChangeRegistration with not supported ttsName (LHPLUS_PHONEMES or SILENCE)
Actual result:
SDL respond with not listed result code UNSUPPORTED_RESOURCE
Expected result:
SDL respond with listed in MOBILE_API.xml result code
Requirements: MOBILE_API.xml

jacobkeeler, 2018-03-08:
I don't believe that these `resultCode` lists within individual RPCs are strictly enforced, but `UNSUPPORTED_RESOURCE` should be included as a potential results for the sake of clarity.

iCollin, 2019-10-22:
Closed via #3091

PR #3091 (2019-10-22): ChangeRegistration can respond with result = UNSUPPORTED_RESOURCE
Fixes #1694

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Manually

### Summary
The ChangeRegistration Response in the MOBILE_API lists a few different possible resultCodes that can be returned but does not list UNSUPPORTED_RESOURCE as one even though this can be returned. (by sending an unsupported ttsName for example)

### Changelog
##### Enhancements
* Makes it more clear to developers that UNSUPPORTED_RESOURCE can be returned from ChangeRegistration

### CLA
- [x] I have s [... cut]
closed 2019-10-22 by iCollin