smartdevicelink/sdl_core #1709: Mandatory parameters are not sent in SDL.ActivateApp response when EnablePolicy=false

jacobkeeler, 2017-08-07:
From Slack `core-help` channel:

> 1. SDL.ActivateApp RPC response is having below parameters
> - isPermissionsConsentNeeded as boolean and mandatory
> - isAppPermissionsRevoked as boolean and mandatory
> - appRevokedPermissions as Common.PermissionItem array and not mandatory
> - isAppRevoked as boolean and mandatory
> 
> 2. If policy is disabled in smartDeviceLink.ini as "EnablePolicy = false" then none of the above mentioned parameters are coming from SDL to HMI. It just gives "code and method" as mentioned below.
> 
> Debug: SDL [application_manager_impl.cc:1999] : Convertion result: { "id" : 10, "jsonrpc" : "2.0", "result" : { "code" : 0, "method" : "SDL.ActivateApp" } }

TiwDeepak, 2017-09-11:
This is unnecessary check in policy_handler.cc 
void PolicyHandler::OnActivateApp(uint32_t connection_key,
                                  uint32_t correlation_id) {
  LOG4CXX_AUTO_TRACE(logger_);
  POLICY_LIB_CHECK_VOID(); =>>> Comment this check and it will work properly ..
  ApplicationSharedPtr app = application_manager_.application(connection_key);
  if (!app.valid()) {
    LOG4CXX_WARN(logger_, "Activated App failed: no app found.");
    return;
  }
  std::string policy_app_id = app->policy_app_id();

  AppPermissions permissions(policy_app_id);

  sync_primitives::AutoReadLock lock(policy_manager_lock_);
  if (!policy_manager_) {
    LOG4CXX_DEBUG(logger_, "The shared library of policy is not loaded");
    if (!PolicyEnabled()) {
      permissions.isSDLAllowed = true;
    }
  }

iCollin, 2021-03-31:
Closed as not reproducible against the first 7.1 release candidate.

closed 2021-03-31 by iCollin