smartdevicelink/sdl_core #1717: Build fails on latest develop when triggered second time

shoamano83, 2017-08-12:
### Step to reproduce:
1. Check out latest develop branch (commit 4e7e56009daf055d3a5ef51d1d5c1587417390c2)
2. Apply fix of PR https://github.com/smartdevicelink/sdl_core/pull/1716
2. create a build directory ($ cd .. && mkdir cmake_build && cd cmake_build)
3. Give a build ($ cmake ../sdl_core && make). Type your sudo password if asked
4. After make successfully finished, run 'make' once again

### Expected Result:
SDL Core should build without any issue

### Actual Result:
Build fails with following error message:

```
[ 23%] Performing update step for 'libbson'
Current branch master is up to date.
No stash found.
No stash found.
No stash found.
CMake Error at /home/samano/projects/sdl_core_build/cmake_build/src/3rd_party/libbson-prefix/tmp/libbson-gitupdate.cmake:136 (message):
  

  Failed to unstash changes in:
  '/home/samano/projects/sdl_core_build/cmake_build/src/3rd_party/bson_c_lib/'.

  You will have to resolve the conflicts manually

src/3rd_party/CMakeFiles/libbson.dir/build.make:94: recipe for target 'src/3rd_party/libbson-prefix/src/libbson-stamp/libbson-update' failed
make[2]: *** [src/3rd_party/libbson-prefix/src/libbson-stamp/libbson-update] Error 1
CMakeFiles/Makefile2:295: recipe for target 'src/3rd_party/CMakeFiles/libbson.dir/all' failed
make[1]: *** [src/3rd_party/CMakeFiles/libbson.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

### Tested environment:
Ubuntu 16.04 (x86_64) on a VirtualBox machine

jacobkeeler, 2017-08-18:
Closing with the merge of https://github.com/smartdevicelink/bson_c_lib/pull/3

closed 2017-08-18 by jacobkeeler

shoamano83, 2017-09-07:
I noticed that I still see a build error if the system has old version of git. Created PR: https://github.com/smartdevicelink/bson_c_lib/pull/6