smartdevicelink/sdl_core #1727: RC application does not release allocated resources in case user exit application from HMI

aderiabin, 2017-08-17:
# Description
RC application does not release allocated resources in case user exit application from HMI

## Preconditions
1. Make sure SDL is built with PROPRIETARY flag
2. Start SDL and HMI, first ignition cycle
3. Register 2 mobile applications (App1, App2)
 (App1: NONE, App2: NONE)

## Steps to reproduce
1. Enable RC from HMI with AUTO_ALLOW access mode
2. Activate App1 on HMI
 (App1: FULL, App2: NONE)
3. Send valid SetInteriorVehicleData RPC with CLIMATE module data from App1
4. Activate App2 on HMI
 (App1: LIMITED, App2: FULL)
5. Activate and exit application App1 on HMI (BasicCommunication.OnExitApplication)
 (App1: NONE, App2: LIMITED)
6.  Send valid SetInteriorVehicleData RPC with CLIMATE module data from App2

## Actual result
SDL reject SetInteriorVehicleData RPC and does not send request to HMI. 
Module CLIMATE is still allocated to App1.

## Expected result
SDL process SetInteriorVehicleData RPC and send request to HMI. 
Module CLIMATE is allocated to App2.

# Environment
SDL commit: aec6b1
WebHMI commit: f2b024
OS with SDL: Ubuntu 14.04
VM used for OS with SDL (yes/no): no
Transport: USB

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ] UTs add/update
- [ ] ATF tests add/update
- [ ] Manual tests
- [ ] Add/update CI plans/jobs
- [ ] SDD updates
- [ ] Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides))
- [ ] Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines))

HSavynetska, 2018-07-23:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1983
SDL build: develop branch ( 9727cd2 )
ATF: develop branch

Status: Defect is not reproduced.

mkorniichuk, 2019-09-30:
Since new features have been merged to a develop of SDL Core, the script has to be updated as well. Please, find a new version of the script in the issue description attachments.

The issue is not  reproducible.

SDL Core: 66888d419aeb753d7e323d31dc25e59ce4eab6e7
ATF: [`ddcc82a`](https://github.com/smartdevicelink/sdl_atf/tree/ddcc82a9db42592f92849f4c93696113820e3c86)

closed 2019-09-30 by mkorniichuk