smartdevicelink/sdl_core #1734: [SDL 0083] Expandable Design for Proprietary Data Exchange

theresalech, 2017-08-23:
Proposal: [Expandable Design for Proprietary Data Exchange](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0083-Expandable-design-for-proprietary-data-exchange.md)

> The SystemRequest/OnSystemRequest RPCs provide a mechanism for various components within the head unit to exchange data with app(s) written specifically to communicate with those components. With the growing popularity of SDL as a means of data exchange between head units and the outside world, there is a need to provide service to more such components. Today, the list of such components is limited to the 'RequestType' enum. Expanding this list everytime would not be an ideal solution. This document proposes a new mechanism for an expandable design in such scenarios by introducing a new string field called 'RequestSubType' to be used along with a new 'RequestType' enum called 'OEM_SPECIFIC'.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/249

Steering Committee Decision:

> When discussing this proposal, there was an ask for an example of the data that would be exchanged. Ford provided the example of an embedded navigation getting traffic information from a companion app. The Steering Committee then voted to accept this proposal.

yang1070, 2018-05-14:
@theresalech Ford has tested and reviewed this Delivery. And it is ready for Livio to review.

SDL/   Suplimentary Tool | GitHub   branch | GitHub   PR | Commit | Comments
-- | -- | -- | -- | --
SDL | sdl_core/tree/feature/expandable_design_for_proprietary_data_exchange (https://github.com/smartdevicelink/sdl_core/tree/feature/expandable_design_for_proprietary_data_exchange) | PR#2149 | 14e4c35 | This branch contains   SDL Core implementation.
WebHMI | sdl_hmi/tree/feature/expandable_design_for_proprietary_data_exchange (https://github.com/smartdevicelink/sdl_hmi/tree/feature/expandable_design_for_proprietary_data_exchange) | PR#73 | 84864dd | This branch contains Web HMI   implementation.
ATF scripts | sdl_atf_test_scripts/tree/feature/expandable_design_for_proprietary_data_exchange (https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/feature/expandable_design_for_proprietary_data_exchange)  | PR#1877 | b9be178 | This branch contains ATF Test scripts.
ATF | sdl_atf/tree/develop (https://github.com/smartdevicelink/sdl_atf/tree/develop) |   | 836a8dc | This branch contains the Automated Test Framework tool
HMI Integration Guidelines | sdl_hmi_integration_guidelines/tree/feature/expandable_design_for_proprietary_data_exchange (https://github.com/smartdevicelink/sdl_hmi_integration_guidelines/tree/feature/expandable_design_for_proprietary_data_exchange)  | PR#98 | e6655bf | This branch contains SDL HMI Integration Guidelines

theresalech, 2018-05-14:
@yang1070 thank you!  Can you please tag me in the PRs that have been reviewed and approved by Ford?

jacobkeeler, 2018-06-11:
Closing with the merge of #2149

PR #2127 (2018-04-19): Initial implementation of Expandable Design for Proprietary Data Exchange
Fixes #1734 

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
ATF test scripts will be provided

### Summary
- Made the following changes to Mobile API:
  Added new parameter "requestSubType" to SystemRequest and
  OnSystemRequest notification
  Added new RequestType - "OEM_SPECIFIC"

- Made the following changes to HMI API:
Added new parameter "requestSubType" to HMIApplication structure
and to OnAppPermissionChange notification
Added new RequestType - "OEM_SPECIFIC"

-  [... cut]
PR #2185 (2018-05-15): Feature/expandable design for proprietary data exchange external policy
Fixes #1734 

This PR is **ready** for review.

### Risk
This PR makes **NO** API changes.

### Testing Plan
Unit tests are provided
ATF test scripts are provided

### Summary
Implemented support for EXTERNAL_PROPRIETARY flow
### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #2149 (2018-06-06): Feature/expandable design for proprietary data exchange
Fixes #1734 

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Unit tests are provided
ATF test scripts are provided

### Summary
- Made the following changes to Mobile API:
  Added new parameter "requestSubType" to SystemRequest and
  OnSystemRequest notification
  Added new RequestType - "OEM_SPECIFIC"

- Made the following changes to HMI API:
Added new parameter "requestSubType" to HMIApplication structure
and to OnAppPermissionChange notification
Added new RequestType - "OEM_SPECIFIC"

 [... cut]
closed 2018-06-11 by jacobkeeler