smartdevicelink/sdl_core #1736: DCHECK_OR_RETURN_VOID() ambiguous overload for 'operator==' issue

phongt, 2017-08-24:
There are some unconditional usage of DCHECK_XX() macro. It causes the compilation error.
eg: 
```In file included from /home/phongtran/Works/AGL/master/tmp/work/i586-poky-linux/sdl-core/4.1.0+gitAUTOINC+5043a62c8e-r0/git/src/components/include/utils/shared_ptr.h:36:0,
|                  from /home/phongtran/Works/AGL/master/tmp/work/i586-poky-linux/sdl-core/4.1.0+gitAUTOINC+5043a62c8e-r0/git/src/components/include/utils/threads/message_loop_thread.h:43,
|                  from /home/phongtran/Works/AGL/master/tmp/work/i586-poky-linux/sdl-core/4.1.0+gitAUTOINC+5043a62c8e-r0/git/src/components/protocol_handler/include/protocol_handler/protocol_handler_impl.h:44,
|                  from /home/phongtran/Works/AGL/master/tmp/work/i586-poky-linux/sdl-core/4.1.0+gitAUTOINC+5043a62c8e-r0/git/src/components/protocol_handler/src/protocol_handler_impl.cc:33:
| /home/phongtran/Works/AGL/master/tmp/work/i586-poky-linux/sdl-core/4.1.0+gitAUTOINC+5043a62c8e-r0/git/src/components/protocol_handler/src/protocol_handler_impl.cc: In member function 'virtual void protocol_handler::ProtocolHandlerImpl::OnTMMessageSendFailed(const transport_manager::DataSendError&, protocol_handler::RawMessagePtr)':
| /home/phongtran/Works/AGL/master/tmp/work/i586-poky-linux/sdl-core/4.1.0+gitAUTOINC+5043a62c8e-r0/git/src/components/include/utils/macro.h:66:5: error: ambiguous overload for 'operator==' (operand types are 'const RawMessagePtr {aka const utils::SharedPtr<protocol_handler::RawMessage>}' and 'int')
|      assert(condition);    \
|      ^
```
More details can find  in the attach [log.do_compile.txt](https://github.com/smartdevicelink/sdl_core/files/1250075/log.do_compile.txt)
This compile error happen in Yocto environment (gcc version 7.2.0 (GCC))

LuxoftAKutsan, 2017-08-29:
@phongt is it develop sdl or master?

phongt, 2017-08-29:
@LuxoftAKutsan release/4.1_LTS branch

phongt, 2017-08-30:
Change to use the condition expression in DCHECK_XX() macro. The protocol_handler_impl.cc can compile. 
For example:
diff --git a/src/components/protocol_handler/src/protocol_handler_impl.cc b/src/components/protocol_handler/src/protocol_handler_impl.cc
index aa29107..aca59c2 100644
--- a/src/components/protocol_handler/src/protocol_handler_impl.cc
+++ b/src/components/protocol_handler/src/protocol_handler_impl.cc
@@ -628,7 +628,7 @@ void ProtocolHandlerImpl::OnTMMessageSend(const RawMessagePtr message) {
 void ProtocolHandlerImpl::OnTMMessageSendFailed(
     const transport_manager::DataSendError& error,
     const RawMessagePtr message) {
-  DCHECK_OR_RETURN_VOID(message);
+  DCHECK_OR_RETURN_VOID(!(message == nullptr));
   // TODO(PV): implement
   LOG4CXX_ERROR(logger_,
                 "Sending message " << message->data_size()
@@ -805,7 +805,7 @@ RESULT_CODE ProtocolHandlerImpl::SendMultiFrameMessage(
 }
 
 RESULT_CODE ProtocolHandlerImpl::HandleMessage(const ProtocolFramePtr packet) {
-  DCHECK_OR_RETURN(packet, RESULT_UNKNOWN);
+  DCHECK_OR_RETURN(!(packet == nullptr), RESULT_UNKNOWN);
   LOG4CXX_DEBUG(logger_, "Handling message " << packet);
   switch (packet->frame_type()) {
     case FRAME_TYPE_CONTROL:
@@ -1196,7 +1196,7 @@ void ProtocolHandlerImpl::PopValideAndExpirateMultiframes() {
        it != frame_list.end();
        ++it) {
     const ProtocolFramePtr frame = *it;
-    DCHECK(frame);
+    DCHECK(!(frame == nullptr));
     if (!frame) {
       continue;
     }

LuxoftAKutsan, 2017-09-01:
@phongt  you use nullptr that is C++11. SDL currently use C++98.

phongt, 2017-09-01:
@LuxoftAKutsan  you are correct.  This only wants to show that the parameter of DCHECK_XX() should be a condition expression. 
Maybe the compiler options in Yocto environment is strictly. So the compile errors happen.

phongt, 2017-09-01:
@LuxoftAKutsan For your information, [#L187](https://github.com/smartdevicelink/sdl_core/blob/release/4.1_LTS/CMakeLists.txt#L187) C++ -std is gnu++0x currently.  So nullptr can compile with it.

LuxoftAKutsan, 2017-09-05:
> @LuxoftAKutsan For your information, #L187 C++ -std is gnu++0x currently. So nullptr can compile with it.
@phongt t I know, but that is but but not a feature. 
Soon I we will move SDL to newer compiler and c++ version, it should be specified in README, and checked all functionality of SDL on certain compiler version.

chbae, 2017-12-24:
I'm upgrading yocto to rocko in AGL and I got the same errors in release/4.4.1 branch.

jacobkeeler, 2022-01-14:
@phongt Are you still experiencing this issue? We moved this project on to C++11 `shared_ptr`s quite a while back, which might have fixed the issue

phongt, 2022-01-14:
@jacobkeeler  I don't follow this for a long time. I would close it now.