smartdevicelink/sdl_core #1737: Dead lock in toHMI or fromHMI Message loop threads- Resulting SDL core not responding HMI requests

madhuy, 2017-08-25:
We noticed Deadlock in toHMI or fromHMI Message loop threads when we run in with more apps.
Repro:
1. Install 20+ apps in Android HS - Note if you have more apps it is consistently reproducible
2. start SDL core and HMI
Linux Version  - 3.10.31-ltsi

**My analysis on logs in case of toHMI dead lock**

_1. Thread created with id **3045880800 (0xb58c73e0)**_

`DEBUG [01 Jan 2013 13:19:14,922][Utils][0xb6204220][/data/sdl_core/src/components/utils/src/threads/posix_thread.cc(242)] Thread AM ToHMI #3045880800 started. pthread_result = 0
`

_2. Last Message process_ 

`
DEBUG [01 Jan 2013 13:19:37,395][ApplicationManager][0xb58d73e0][/data/sdl_core/src/components/application_manager/src/application_manager_impl.cc(2088)] Converted message, trying to create hmi command
`

_3. Below toHMI messages not processed SDL core is not responding back to HMI though it recieves from HMI - May be dead lock . _
`
INFO  [01 Jan 2013 13:19:38,047][HMIMessageHandler][0xb61033e0][/data/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc(115)] Received message from hmi
TRACE [01 Jan 2013 13:19:38,047][ApplicationManager][0xb61033e0][/data/sdl_core/src/components/application_manager/src/application_manager_impl.cc(829)] Enter
TRACE [01 Jan 2013 13:19:38,047][ApplicationManager][0xb61033e0][/data/sdl_core/src/components/application_manager/src/application_manager_impl.cc(829)] Exit
INFO  [01 Jan 2013 13:19:38,047][HMIMessageHandler][0xb61033e0][/data/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc(123)] Message from hmi given away.
`

_4. Last message from the toHMI thread (0xb58c73e0 - 3045880800)_

` 
TRACE [01 Jan 2013 13:19:45,333][ApplicationManager][0xb58c73e0][/data/sdl_core/src/components/application_manager/src/application_manager_impl.cc(2694)] Exit
`

_5.  Message toHMI not processed . SDL core not responding to HMI requests- May be dead lock or Thread died_

`
INFO  [01 Jan 2013 13:19:45,448][HMIMessageHandler][0xb61033e0][/data/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc(123)] Message from hmi given away.
TRACE [01 Jan 2013 13:19:45,448][HMIMessageHandler][0xb4eb73e0][/data/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc(61)] Exit
INFO  [01 Jan 2013 13:19:45,448][HMIMessageHandler][0xb4eb73e0][/data/sdl_core/src/components/hmi_message_handler/src/messagebroker_adapter.cc(182)] Successfully sent to observer
TRACE [01 Jan 2013 13:19:45,448][HMIMessageHandler][0xb4eb73e0][/data/sdl_core/src/components/hmi_message_handler/src/messagebroker_adapter.cc(152)] Exit
TRACE [01 Jan 2013 13:19:45,448][HMIMessageHandler][0xb4eb73e0][/data/sdl_core/src/components/hmi_message_handler/src/messagebroker_adapter.cc(85)] Exit
INFO  [01 Jan 2013 13:19:45,448][HMIMessageHandler][0xb61033e0][/data/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc(115)] Received message from hmi
TRACE [01 Jan 2013 13:19:45,449][ApplicationManager][0xb61033e0][/data/sdl_core/src/components/application_manager/src/application_manager_impl.cc(829)] Enter
TRACE [01 Jan 2013 13:19:45,449][ApplicationManager][0xb61033e0][/data/sdl_core/src/components/application_manager/src/application_manager_impl.cc(829)] Exit
INFO  [01 Jan 2013 13:19:45,449][HMIMessageHandler][0xb61033e0][/data/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc(123)] Message from hmi given away.
`

Attcahing full logs [smartdevicelink_toHMI_deadlock.txt](https://github.com/smartdevicelink/sdl_core/files/1253238/smartdevicelink_toHMI_deadlock.txt)

Note:
 We did experiment by increasing thread min stack size to 65536 from 16k. Issue is still reproducible.

Let us know if you need more info.

JackLivio, 2017-08-28:
@madhuy Are you using Master or Develop?

JackLivio, 2017-08-28:
I tried with 25 apps registered at the same time, and then registered 25 more while activating each one after registration. I would test a SEEK_RIGHT in each app and the apps continued to receive the message from the HMI. I tried this on Master and Develop. I am on ubuntu 14 using the default configurations.

madhuy, 2017-08-28:
@JackLivio , Thank you quick response.
We are using https://github.com/smartdevicelink/sdl_core/tree/release/4.1_LTS
Attaching more logs [sdlcore_1737_deadlock_fromHMI.zip] which contain Message broker logsto prove that it is not websocket client issue (https://github.com/smartdevicelink/sdl_core/files/1257654/sdlcore_1737_deadlock_fromHMI.zip)

**Last successfully handled RPC is "id":107**

`
TRACE [01 Jan 2013 16:59:23,146][ApplicationManager][0xb55ff3e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/application_manager_impl.cc(2688)] Enter
TRACE [01 Jan 2013 16:59:23,146][TransportManager][0xb5eff3e0][/home/sdl/host_dir/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc(616)] Exit
DEBUG [01 Jan 2013 16:59:23,146][TransportManager][0xb5eff3e0][/home/sdl/host_dir/sdl_core/src/components/transport_manager/src/transport_manager_impl.cc(917)] Got adapter 0x12be58[6] by session id 1
DEBUG [01 Jan 2013 16:59:23,146][#19400][0xb55ff3e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/message.cc(215)] TID: 833	(virtual void application_manager::ApplicationManagerImpl::Handle(application_manager::impl::MessageFromHmi)@/home/sdl/host_dir/sdl_core/src/components/application_manager/src/application_manager_impl.cc:2695)	From HMI	function_id: 0, correlation_id: 0, connection_key: 0, message_type: -1, protocol_version: 0, data_size: 0, payload_size: 0, binary_data: 0, json_message: {"id":107,"jsonrpc":"2.0","result":{"code":0,"method":"UI.AddCommand"}}

TRACE [01 Jan 2013 16:59:23,146][TransportManager][0xb5eff3e0][/home/sdl/host_dir/sdl_core/src/components/transport_manager/src/transport_adapter/transport_adapter_impl.cc(302)] enter. device_id: 0xb5b02614, app_handle: 0xb5b02618, data: 1
TRACE [01 Jan 2013 16:59:23,146][ApplicationManager][0xb55ff3e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/application_manager_impl.cc(2071)] Enter
TRACE [01 Jan 2013 16:59:23,146][ApplicationManager][0xb55ff3e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/application_manager_impl.cc(1758)] Enter
TRACE [01 Jan 2013 16:59:23,146][TransportManager][0xb5eff3e0][/home/sdl/host_dir/sdl_core/src/components/transport_manager/src/transport_adapter/transport_adapter_impl.cc(900)] enter. device_id: 0xb5b02614, app_handle: 0xb5b02618
DEBUG [01 Jan 2013 16:59:23,146][ApplicationManager][0xb55ff3e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/application_manager_impl.cc(1762)] 			Message to convert: protocol 0; json {"id":107,"jsonrpc":"2.0","result":{"code":0,"method":"UI.AddCommand"}}
`

**Failure to process RPC “id”:108 as below logs**

`DEBUG [01 Jan 2013 16:59:23,181][HMIMessageHandler][0xb60f03e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/message.cc(215)] TID: 838    (virtual void hmi_message_handler::HMIMessageHandlerImpl::Handle(hmi_message_handler::impl::MessageFromHmi)@/home/sdl/host_dir/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc:122)    HMIMessageHandler    function_id: 0, correlation_id: 0, connection_key: 0, message_type: -1, protocol_version: 0, data_size: 0, payload_size: 0, binary_data: 0, json_message: {“id”:108,“jsonrpc”:“2.0”,“result”:{“code”:0,“method”:“UI.AddCommand”}}
`
Between id's 107 and 108 "HMI levels timer" is restarted.

Could you please analyze logs and provide me feedback.

madhuy, 2017-08-28:
attaching one more set of logs
[SDlCore_1747_deadlock_fromHMI_1.zip](https://github.com/smartdevicelink/sdl_core/files/1257682/SDlCore_1747_deadlock_fromHMI_1.zip)
in  this log **id:94** succeed and **id:95** fail to process

**Success case**
`
DEBUG [01 Jan 2013 10:19:06,023][ApplicationManager][0xb55ff3e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/application_manager_impl.cc(1762)] 			Message to convert: protocol 0; json {"id":94,"jsonrpc":"2.0","result":{"code":0,"method":"UI.Show"}}
`

**Failure to proecss below RPC**
`
DEBUG [01 Jan 2013 10:19:06,044][HMIMessageHandler][0xb60f63e0][/home/sdl/host_dir/sdl_core/src/components/application_manager/src/message.cc(215)] TID: 831	(virtual void hmi_message_handler::HMIMessageHandlerImpl::Handle(hmi_message_handler::impl::MessageFromHmi)@/home/sdl/host_dir/sdl_core/src/components/hmi_message_handler/src/hmi_message_handler_impl.cc:122)	HMIMessageHandler	function_id: 0, correlation_id: 0, connection_key: 0, message_type: -1, protocol_version: 0, data_size: 0, payload_size: 0, binary_data: 0, json_message: {"id":95,"jsonrpc":"2.0","result":{"code":0,"method":"UI.SetDisplayLayout"}}
`

JackLivio, 2017-08-28:
I tried again on 4.1LTS with 50 apps that all registered at once, still unable to reproduce the issue.

fronneburg, 2017-08-28:
The problem here is a deadlock between the "ApplicationManagerImpl" "applications_list_lock_" and the "EventDispatcher" "observer_lock_".

If the "Thread Pool 0" thread is processing "ApplicationManagerImpl::OnApplicationRegistered", it AutoLock's the "applications_list_lock_", and then calls "event.raise(event_dispatcher())". This eventually calls "EventDispatcher::RaiseEvent", which tries to acquire the "observer_lock_".

In the meantime, the "AM FromHMI" thread calls "event.raise(application_manager_.event_dispatcher())" from any number of "Run" methods. These eventually call "EventDispatcher::RaiseEvent", acquire the "observer_lock_", and proceed to call the appropriate "on_event" method. Many of these call "application_manager_.application", which tries to acquire the "applications_list_lock_", which is already locked in the "Thread Pool 0" thread.

Result: DEADLOCK.

JackLivio, 2017-08-29:
merged

JackLivio, 2017-08-29:
@madhuy If this exists 4.1LTS could you open another PR for the fix I just merged into develop but against the 4.1LTS branch?

jacobkeeler, 2018-03-07:
Fixed in `4.1_LTS` with the merge of https://github.com/smartdevicelink/sdl_core/pull/1745

closed 2018-03-07 by jacobkeeler