smartdevicelink/sdl_core #1753: Core cannot find App via BT

BrandonHe, 2017-09-11:
### Precondition:
- Ubuntu 16.04 x86 
- Core version 4.3.1
- SPT (Proxy: 20170804-Nightly-Android)
- Android Phone (Android 6.0)
### Reproduce:
1. Run SDL core and SDL_HMI
2. Open SPT on the phone
3. Connect Core and SPT via Bluetooth
4. SPT successfully registered in the HMI
5. Exit the Core
6. Kill the SPT 
7. Restart the Core
8. Re-open SPT on the phone
9. Connect Core and SPT via Bluetooth
10. Connect failed, SPT cannot registered in the HMI
### Probability
**100%**
### Analysis
While core is re-started, it will read BT data from *app_info.dat* file to get the address and channel information. But SPT is not launched, therefore, the connection fail.
```
   "TransportManager" : {
      "BluetoothAdapter" : {
         "devices" : [
            {
               "address" : "84:9F:B5:99:1B:EC",
               "applications" : [
                  {
                     "rfcomm_channel" : "5"
                  },
                  {
                     "rfcomm_channel" : "6"
                  },
                  {
                     "rfcomm_channel" : "7"
                  }
               ],
               "name" : "DIG-AL00"
            }
         ]
      },
      "TcpAdapter" : null
   },
```
Also, the ```void ThreadedSocketConnection::Transmit()``` is blocked in the [poll](https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/transport_manager/src/transport_adapter/threaded_socket_connection.cc#L231), leads the ```Abort()``` function is not implemented. 
So after SPT is launched again, the core ```TransportAdapterImpl::Connect()``` function is detected status as ```ALREADY_EXISTS```

ilikegithubcom, 2017-09-11:
If the connection fails, in the ThreadedSocketConnection:: threadMain () function, you should skip the Transmit function and call the Finalize function to release the connection object.

yangbiao01, 2017-09-11:
Reviewed

jacobkeeler, 2017-10-03:
Fixed for the case originally described in the issue, but not another related case:

### Environment:
- Ubuntu 14.04 x86_64 
- Core branch 4.4.0
- SPT (Proxy: 20170921-Nightly-Android)
- Android Phone (Android 6.0.1)
### Steps to Reproduce:
1. Run SDL core and SDL_HMI
2. Open SPT on the phone
3. Connect Core and SPT via Bluetooth
4. SPT successfully registered in the HMI
5. Exit Core
6. **Restart Core**
7. **Kill the SPT**
8. Re-open SPT on the phone
9. Connect Core and SPT via Bluetooth
10. Connect failed, SPT is unable to register in the HMI
### Probability
**100%**

jacobkeeler, 2017-10-03:
Never mind, it appears that this second scenario has been fixed. Possibly with the merge of https://github.com/smartdevicelink/sdl_core/pull/1791

closed 2017-10-03 by jacobkeeler