smartdevicelink/sdl_core #1768: Core still sends out deprecated Service Data ACK frames

shoamano83, 2017-09-22:
### Description:

SDL Core still sends out Service Data ACK frames every time it receives a media frame from mobile.

According to https://github.com/smartdevicelink/protocol_spec, the frame is marked as `Deprecated`.

Handling and sending such frames will increase CPU load of Core as well as consume a little bit of network bandwidth, especially under high-bitrate video streaming. I think we should remove handling such frames.

shoamano83, 2017-09-22:
We had some discussions with @jacobkeeler and @khburdette on #core-help Slack channel.
- @jacobkeeler heard of issues where the frame was slowing down video streaming.
- According to @khburdette, the original intention of the frame was to give the app info on video frame consumption (allowing the app to speed up or slow down video packets). Since the frame is now meant for ACK, it is kind of useless.
- It was proposed to send out the frames for versions that don’t deprecate them, since old Proxy might be using them.
- So far, we do not know in which protocol version the frame has been deprecated. Description of the frame is added to the document here https://github.com/smartdevicelink/protocol_spec/commit/93c6cccc628327178d7aed89f3ef292bf5cfb08d#diff-04c6e90faac2675aa89e2176d2eec7d8R246 and it says “Deprecated”.

shoamano83, 2017-09-22:
Hi @jacobkeeler, I created a simple PR to limit sending the frame only for protocol versions 2 through 4.

I am not sure whether I can just go on to ask for a review, or we need some more discussion in SDLC over a proposal. Please kindly share your thought. Thanks.

jacobkeeler, 2017-10-03:
Closing with the merge of #1769

closed 2017-10-03 by jacobkeeler