smartdevicelink/sdl_core #1772: Default app policies are never updated after a PTU

AKalinich-Luxoft, 2017-09-22:
**Steps to reproduce**
1. Edit PTU - Add "Location-1" group to default permissions
2. Connect mobile app and bring to FULL
3. Send GetVehicleData(gps) - expected DISALLOWED - OK
4. Perform PTU sequence using PT edited at step 1
5. After successful PTU send GetVehicleData(gps) again

**Expected result**
SDL allows GetVehicleData(gps) request by policy and sends it to HMI

**Actual result**
SDL responds with DISALLOWED and not sends request to HMI

jacobkeeler, 2017-10-03:
Needs to be fixed for PROPRIETARY mode

1. Add Location-1 to the functional_group list in the default app policy on the PTU server
2. Connect test app
3. Perform PTU (app id is not present in update)
4. Bring test app to FULL
5. Send GetVehicleData(gps:true)
6. Get DISALLOWED response

The test app used is left with an outdated version of the `default` permissions in this case. Any new app that is connected will have the updated permissions.

jacobkeeler, 2017-10-03:
App policies in
1. Local PT:
```
        "app_policies": {
            "default": {
                "keep_context": false,
                "steal_focus": false,
                "priority": "NONE",
                "default_hmi": "NONE",
                "groups": ["Base-4", "RemoteControl"],
		"moduleType": [
			"RADIO",
			"CLIMATE"
		]
            },
            "device": {
                "keep_context": false,
                "steal_focus": false,
                "priority": "NONE",
                "default_hmi": "NONE",
                "groups": ["DataConsent-2"]
            },
            "pre_DataConsent": {
                "keep_context": false,
                "steal_focus": false,
                "priority": "NONE",
                "default_hmi": "NONE",
                "groups": ["BaseBeforeDataConsent"]
            }
        }
```
2. Policy Server:
```
        "app_policies": {
            "default": {
                "keep_context": false,
                "steal_focus": false,
                "priority": "NONE",
                "default_hmi": "NONE",
                "groups": ["Base-4", "Location-1"]
            },
            "device": {
                "keep_context": false,
                "steal_focus": false,
                "priority": "NONE",
                "default_hmi": "NONE",
                "groups": ["DataConsent-2"]
            },
            "pre_DataConsent": {
                "keep_context": false,
                "steal_focus": false,
                "priority": "NONE",
                "default_hmi": "NONE",
                "groups": ["BaseBeforeDataConsent"]
            }
        }
```

GetmanetsIrina, 2017-12-15:
Reproduced on SDLOpen commit 9e95f67178c575ca5a537f9ef3a69653219b073d (develop branch)
Fix on https://github.com/AKalinich-Luxoft/sdl_core/pull/1,
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1772_update_default_section.lua.

Reproducing scenario:
1. SDL is built with PROPRIETARY flag
2. Start SDL and HMI, first ignition cycle
3. Device is connected
1. Register new application
2. Perform PTU with updated default section (in script added new group in default section with GetVehicleData RPC)
3. App sends GetVD
Expected result:
SDL performs update and sends OnPermissionsChange to mobile app with updated permissions.
DB has updated info. SDL process GetVD successfully.

mrapitis, 2017-12-21:
Reviewed ATF script and ran successfully.   Successfully reproduced issue with provided commit. Reviewed code with and requested any changes to public methods require appropriate deprecation.

JackLivio, 2018-01-16:
fix merged

PR #1966 (2018-01-16): Fix PTU apply for applications with default permissions
Fixes #1772

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Changes in PR are already covered within existing unit tests.

### Summary
There was a problem in PROPRIETARY flow that SDL does not update permissions for applications with default policies after PTU if application functional group was updated indirectly (through updating of "default" section).

Also SDL does not send OnPermissionChange notification in case described above.

This fix completes fix from #1779

### Changelog
* Added  [... cut]
closed 2018-01-16 by JackLivio