smartdevicelink/sdl_core #1782: Do not allow streaming state change after the app lost approval

t-yoshii, 2017-09-26:
**Description:**

In current implementation, OnVideoDataStreaming notification may be sent by app which already lost streaming approval. This can cause unexpected streaming stop.

Scenario:
When video app1 is switched to video app2, app1's OnVideoDataStreaming(false)/StopStream and app2's StartStream/OnVideoDataStreaming(true) are sent to HMI almost same timing.
If app1's OnVideoDataStreaming(false)/StopStream are sent after app2's StartStream/OnVideoDataStreaming(true), the video on HMI is stopped due to OnVideoDataStreaming(false).

I will create PR not to send OnVideoDataStreaming after app lost approval.

E-SAITO-TMC, 2020-04-15:
@joeygrover -san
A fix has been proposed for this issue. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDL Core, so we believe it is necessary to modify this issue.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives. This particular issue has not yet been raised as high priority. However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.

Please confirm if this is priority for Toyota. If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June. Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-07-27:
Hello, this issue was fixed by https://github.com/smartdevicelink/sdl_core/pull/3154 and is tested by the first test in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2298, which is passing against current develop. Closing as it is no longer reproducible.

closed 2020-07-27 by iCollin