smartdevicelink/sdl_core #1788: build break on develop branch with ENABLE_LOG=OFF

shoamano83, 2017-09-28:
## Steps to reproduce:

1. Check out latest develop branch (commit da4b5766ca3f1fca4caa4b4815073f72dfe37182)
2. Turn off ENABLE_LOG in CMakeLists.txt
3. Give a build

## Result:

I get three build errors:
```
[ 27%] Building CXX object src/components/policy/policy_regular/CMakeFiles/Policy.dir/src/cache_manager.cc.o
/home/samano/projects/sdl_core/sdl_core/src/components/policy/policy_regular/src/cache_manager.cc: In member function ‘virtual void policy::CacheManager::GetUpdateUrls(uint32_t, policy::EndpointUrls&)’:
/home/samano/projects/sdl_core/sdl_core/src/components/policy/policy_regular/src/cache_manager.cc:696:21: error: aggregate ‘std::stringstream service_type_stream’ has incomplete type and cannot be defined
   std::stringstream service_type_stream;
                     ^
```
```
[ 28%] Building CXX object src/components/application_manager/CMakeFiles/AMPolicyLibrary.dir/src/policies/policy_handler.cc.o
/home/samano/projects/sdl_core/sdl_core/src/components/application_manager/src/policies/policy_handler.cc: In member function ‘bool policy::PolicyHandler::IsAppSuitableForPolicyUpdate(std::vector<utils::SharedPtr<application_manager::Application> >::value_type) const’:
/home/samano/projects/sdl_core/sdl_core/src/components/application_manager/src/policies/policy_handler.cc:811:18: error: unused variable ‘app_id’ [-Werror=unused-variable]
   const uint32_t app_id = value->app_id();
                  ^
cc1plus: all warnings being treated as errors
```
```
[ 33%] Building CXX object src/components/application_manager/CMakeFiles/ApplicationManager.dir/src/application_manager_impl.cc.o
/home/samano/projects/sdl_core/sdl_core/src/components/application_manager/src/application_manager_impl.cc: In member function ‘virtual void application_manager::ApplicationManagerImpl::SendMessageToMobile(application_manager::commands::MessageSharedPtr, bool)’:
/home/samano/projects/sdl_core/sdl_core/src/components/application_manager/src/application_manager_impl.cc:1614:18: error: unused variable ‘is_for_plugin’ [-Werror=unused-variable]
       const bool is_for_plugin = plugin_manager_.IsAppForPlugins(app);
                  ^
cc1plus: all warnings being treated as errors
```

luwanjia, 2017-09-28:
I got the same errors. Just modify like this:
----------------------------------------------------------
`diff --git a/CMakeLists.txt b/CMakeLists.txt
index a813b98..8bcd9e0 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -47,7 +47,7 @@ option(BUILD_USB_SUPPORT "libusb support" ON)
 option(BUILD_BACKTRACE_SUPPORT "backtrace support" ON)
 option(BUILD_TESTS "Possibility to build and run tests" OFF)
 option(TELEMETRY_MONITOR "Enable profiling time test util" ON)
-option(ENABLE_LOG "Logging feature" ON)
+option(ENABLE_LOG "Logging feature" OFF)
 option(ENABLE_GCOV "gcov code coverage feature" OFF)
 option(ENABLE_SANITIZE "Sanitize tool" OFF)
 option(ENABLE_SECURITY "Security Ford protocol protection" ON)
diff --git a/src/components/application_manager/src/policies/policy_handler.cc b/src/components/application_manager/src/policies/policy_handler.cc
index ec8b364..7f120ae 100644
--- a/src/components/application_manager/src/policies/policy_handler.cc
+++ b/src/components/application_manager/src/policies/policy_handler.cc
@@ -799,7 +799,9 @@ bool PolicyHandler::IsAppSuitableForPolicyUpdate(
     const Applications::value_type value) const {
   LOG4CXX_AUTO_TRACE(logger_);
 
+#ifdef ENABLE_LOG
   const uint32_t app_id = value->app_id();
+#endif
 
   if (!value->IsRegistered()) {
     LOG4CXX_DEBUG(logger_,
diff --git a/src/components/policy/policy_regular/src/cache_manager.cc b/src/components/policy/policy_regular/src/cache_manager.cc
index 6184320..f89f48c 100644
--- a/src/components/policy/policy_regular/src/cache_manager.cc
+++ b/src/components/policy/policy_regular/src/cache_manager.cc
@@ -36,6 +36,7 @@
 #include <functional>
 #include <ctime>
 #include <cmath>
+#include <sstream>
 
 #include "utils/file_system.h"
 #include "json/reader.h"`
----------------------------------------------------------
It's OK.

jacobkeeler, 2017-10-03:
Closing with the merge of #1789

closed 2017-10-03 by jacobkeeler