smartdevicelink/sdl_core #1798: New remote control modules (LIGHT, AUDIO, HMI_SETTINGS) and parameters (SIS Data)

theresalech, 2017-10-02:
Proposal: [New remote control modules (LIGHT, AUDIO, HMI_SETTINGS) and parameters (SIS Data)](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0099-new-remote-control-modules-and-parameters.md)

> SDL remote control provides the ability for mobile applications to control certain components of the vehicle, such as, tune the radio, set the desired temperature or change the fan speed. However, currently only `CLIMATE` and `RADIO` modules are available for remote-control. Mobile application developers need more than basic climate or radio control. To meet developers' requirements, we propose to add three new modules (`AUDIO`, `LIGHT`, `HMI_SETTINGS`) and some new parameters (heatedXXXEnable and radio sisData) to the existing `CLIMATE` and `RADIO` modules respectively. This proposal only defines new data types and new parameters added to the existing data types. It does not add or change any remote control RPCs.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/296

Steering Committee Decision:

> The Steering Committee voted to accept this proposal with revisions. The author will revise the proposal to fix a few typos, as well as include descriptions for the groups of lights - listing which single light elements they typically include. It was also noted that limiting the scope of proposals, especially those related to Remote Control, in the future will help to make for easier understanding and discussion among SDLC Members.
>Proposal has now been revised to include the descriptions and fix the typos mentioned.

theresalech, 2018-07-11:
Note that this feature is revised by [SDL 0182](https://github.com/smartdevicelink/sdl_core/issues/2338)

theresalech, 2018-08-13:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [Revise SDL-0099 - New remote control modules](https://github.com/smartdevicelink/sdl_evolution/pull/572).  Implementation will need to be revised to remove the duplicate structs and reuse `GPSData` and `RGBColor`.

jacobkeeler, 2018-08-24:
Closing with the merge of #2207

closed 2018-08-24 by jacobkeeler

jordynmackool, 2018-10-03:
Proposal file has been updated per the revisions in  [this accepted PR](https://github.com/smartdevicelink/sdl_evolution/pull/601/files).

theresalech, 2018-10-10:
Proposal file has been updated per the revisions in  [this accepted PR](https://github.com/smartdevicelink/sdl_evolution/pull/601/files).  Core PR #2207 reflects revisions.