smartdevicelink/sdl_core #1813: Regarding update to HMI documentation

VGunasekaranHarman, 2017-10-11:
Hi,
Have downloaded the HMI documentation from the link - https://d83tozu1c8tt6.cloudfront.net/media/builds/pdf/hmi-documentation-english-e91a369a.pdf

Have been taken to this link from https://smartdevicelink.com/resources/pdfs/

It looks that all the HMI RPCs are not defined in the HMI documentation. For eg, OnEventChanged

Please provide the updated document.

Thanks.

jacobkeeler, 2017-10-12:
I'm not certain what is covered by the document you linked, but you can find the HMI Integration Documentation (including RPCs) [here](https://smartdevicelink.com/en/docs/hmi/master/overview/) on the developer portal

Questions like this are best kept to the Slack group. In addition this is not SDL Core related. please post in the appropriate channel on [Slack](http://slack.smartdevicelink.com/).

closed 2017-10-12 by jacobkeeler

joeygrover, 2017-10-13:
@VGunasekaranHarman since this is specific to the HMI documentation, in the future please raise these issues in the documentation repo. Found here: https://github.com/smartdevicelink/sdl_hmi_integration_guidelines

I have already raised an issue for this specific missing RPC [here](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines/issues/82).