smartdevicelink/sdl_core #1824: building on Qt. Error in src/component/dbus/CmakeLists.txt line #77

shadowQ5, 2017-10-19:
**How to reproduce**
 cmake -DHMI=qt -DHMI2=ON ../path/to/sdl/

**Fix entry**
add_dependecnies -> add_dependencies

jacobkeeler, 2017-10-19:
This issue is known, a fix was submitted (#1730), but the fact is that the QT HMI hasn't been supported in several years. I intend on writing a [SDL Evolution](https://github.com/smartdevicelink/sdl_evolution) proposal in the near future to figure out how the SDLC would like to move forward with the QT HMI.

jacobkeeler, 2018-03-02:
DBUS build will be fixed in the next release (https://github.com/smartdevicelink/sdl_core/pull/2005), but the `-DHMI=qt` option will no longer be available (https://github.com/smartdevicelink/sdl_core/issues/1857).

closed 2018-03-02 by jacobkeeler