smartdevicelink/sdl_core #1831: Crash during Navigation app resume

VGunasekaranHarman, 2017-10-27:
[onActivateAppResponseCrashIssue.zip](https://github.com/smartdevicelink/sdl_core/files/1420920/onActivateAppResponseCrashIssue.zip)

Hi,
We are observing a crash during the Navigation app resume. Version used is SDL 4.2

Below are our current analysis.
1. BC.ActivateApp is sent from SDL to HMI
2. Once the HMI gives the response to the BC.ActivateApp, then there is a crash observed in state_controller_impl.cc where below mentioned DCHECK is failing.

void StateControllerImpl::OnActivateAppResponse(
    const smart_objects::SmartObject& message) {
....
DCHECK_OR_RETURN_VOID(pending_state);
....
}

Please let us know whether any fix is made for this issue in SDL.
Thanks.

JackLivio, 2017-11-20:
@VGunasekaranHarman I am not able to reproduce this issue when using the newest version of SDL core. There might have been a fix for this since 4.2 but that would be difficult to track over the hundreds of commits made since then. Looking at the blame on github for that file, StateControllerImpl::OnActivateAppResponse() hasn't been updated in 2-3 years so the fix might be elsewhere in the project. 

My suggestion is to temporarily test this issue in 4.4 and see if the issue still exists in an effort to rule out environmental issues.

VGunasekaranHarman, 2017-12-05:
Hi JackLivio,

Thanks for the input.
We have verified the problem at our side.
Its due to our deployment model where the Applink layer and the SDL runs in same process.
We observed the issue because of the reason that Applink layer provides response to the RPC even before SDL adds it to pending_state.
I feel that code need to deployment model independent.

Is there any plan to address such issues in future??

Thanks,
Varadha.

Ypostolov, 2018-09-13:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2048
SDL build: develop branch ( 28235f88a6733106a7e8a0d490dfb16a5f374a33 )
ATF: develop branch

Status: Defect is not reproduced in current develop.

iCollin, 2021-03-31:
Closed as not reproducible.
Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin