smartdevicelink/sdl_core #1836: FindGstreamer-1.0.cmake doesn't work correctly on Ubuntu 16.04

jacobkeeler, 2017-11-02:
### Bug Report
Only occurs with Ubuntu 16.04 onward, the gstreamer CMake module is not able to find the gstreamer installation directory. This can be worked around using:

```
export GSTREAMER_DIR=$PATH_TO_GSTREAMER
```

Where PATH_TO_GSTREAMER = /usr/lib/x86_64-linux-gnu/gstreamer-1.0 for Ubuntu 16.04 64-bit.

Related: https://github.com/smartdevicelink/sdl_core/pull/924

##### Reproduction Steps
1. Run `cmake ../sdl_core -DEXTENDED_MEDIA_MODE=ON`

##### Expected Behavior
The gstreamer-1.0 libraries are found

##### Observed Behavior
The gstreamer-1.0 libraries are not found, resulting in the following message:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_gstconfig_INCLUDE_DIR
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04 LTS 64-bit
* SDL Core Version: 4.4.0

aderiabin, 2017-12-01:
Contributor priority is set `Low` with reason: The issue is related to specific behavior of cmake.

JackLivio, 2019-10-29:
Not reproducible on ubuntu 18. 

For anyone experiencing this issue try installing package libgstreamer1.0-dev

closed 2019-10-29 by JackLivio