smartdevicelink/sdl_core #1846: Mismatch for mandatory parameter between MOBILE_API and HMI_API

drauhut, 2017-11-07:
There is a  mismatch for the mandatory parameter between the MOBILE_API and the HMI_API for the following parameters:

- distanceToManeuver
- distanceToManeuverScale

The MOBILE_API is reflecting that these parameters are not mandatory, whereas the HMI_API reflects that they are mandatory.

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to API.

robinmk, 2018-09-27:
@jacobkeeler , would this be considered as a breaking change and would it be okay to fix this as bug?

HSavynetska, 2019-11-22:
The issue is relevant on the current develop branch

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2120
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

jacobkeeler, 2020-10-06:
Closing since proposal was accepted and separate issue was submitted: https://github.com/smartdevicelink/rpc_spec/issues/227

GetmanetsIrina, 2021-04-06:
@jordynmackool, The issue has been reopened because it was not fixed within smartdevicelink/rpc_spec#227 proposal.

`distanceToManeuver` and `distanceToManeuverScale` parameters are still not mandatory in the [Mobile API](https://github.com/smartdevicelink/rpc_spec/blob/6b98355357b5b1893bbb59cb668d28545023457c/MOBILE_API.xml#L7484) and mandatory in [HMI API](https://github.com/smartdevicelink/sdl_core/blob/a0e93bb70e0e3bce3b727e2fb06f39847d3b23bd/src/components/interfaces/HMI_API.xml#L5941)

JackLivio, 2021-04-06:
FYI this is still a breaking change if we want to change the mandatory values in the mobile api / RPC spec. The proposal does not mention changing those values either. The proposal only mentions a description update. 

I believe the proposal requires a revision stating this will be a breaking change to the RPC Spec, and wont be able to complete its implementation until a major version change (Core 8.0).