smartdevicelink/sdl_core #1854: SDL Core dump after connect/disconnect USB cable

GetmanetsIrina, 2017-11-14:
**Occurrence: Rare** 

## Preconditions
- SDL is built with PROPRIETARY flag.
- SDL, HMI are started.

## Steps to reproduce
1. Connect mobile via USB
2. Register app with HMI type "REMOTE_CONTROL"
3. Disconnect mobile via USB

## Actual result
SDL Core Dump
```WARN [11 Jul 2017 11:55:47,424][0x7f70d3fff700][ApplicationManager] /home/astasiuk/GitHub/OPENSDL/sdl_core/src/components/application_manager/src/application_manager_impl.cc:1456 ApplicationManagerImpl::SendMessageToMobile: Function "OnHMIStatus" (#32768) not allowed by policyr```
```
gdb smartDeviceLinkCore core
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from smartDeviceLinkCore...done.

warning: core file may not match specified executable file.
[New LWP 484]
[New LWP 445]
[New LWP 448]
[New LWP 456]
[New LWP 447]
[New LWP 462]
[New LWP 449]
[New LWP 450]
[New LWP 478]
[New LWP 444]
[New LWP 452]
[New LWP 453]
[New LWP 461]
[New LWP 823]
[New LWP 839]
[New LWP 1356]
[New LWP 479]
[New LWP 483]
[New LWP 443]
[New LWP 451]
[New LWP 454]
[New LWP 455]
[New LWP 457]
[New LWP 458]
[New LWP 459]
[New LWP 460]
[New LWP 477]
[New LWP 480]
[New LWP 481]
[New LWP 482]
[New LWP 485]
[New LWP 486]
[New LWP 487]
[New LWP 488]
[New LWP 446]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./smartDeviceLinkCore'.
Program terminated with signal SIGABRT, Aborted.
#0 0x00007f9bea936c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56 ../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb)
```

## Expected result
No SDL core dumps

# Environment
SDL commit: 10b632f27b3e04ed1e75363d0674c44279f7ac1f
OS with SDL: Ubuntu 16.04
VM used for OS with SDL (yes/no): **no**
ATF version: n/a
Transport: USB
HTML5 HMI version: develop(e206fc7b8dc8a7130395da3b42ec9af97d14d4e7)
Mobile device: Sumsung GT-I9190
Mobile OS: Android4.4.2
Mobile App: SyncProxyTester, version:2.1.86

# Attachments
[SmartDeviceLinkCore_usb_unplug.zip](https://github.com/smartdevicelink/sdl_core/files/1471186/SmartDeviceLinkCore_usb_unplug.zip)
[bt_usb_unplug_remote_control.zip](https://github.com/smartdevicelink/sdl_core/files/1471187/bt_usb_unplug_remote_control.zip)
[core.zip](https://github.com/smartdevicelink/sdl_core/files/1471189/core.zip)
[SmartDeviceLinkCore.log.zip](https://github.com/smartdevicelink/sdl_core/files/1471192/SmartDeviceLinkCore.log.zip)

# Expected delivery
- [ ] Source code updates
- [x] **Code comments _(not required)_**
- [x] **UTs add/update _(not required)_**
- [x] **ATF tests add/update _(not required)_**
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to functionality: protocol handling and SDL behavior leads to core crash.

GetmanetsIrina, 2017-12-22:
According to last attached logs there are a problem in UsbConnection class.
OutTransferCallback sometimes called after UsbConnection destructor call and this is a reason of core crash. It is possible when in Finalise() function libusb_cancel_transfer returns error and SDL will not wait for transfer will be finished its work. As possible solution we should also wait until out_transfer_ will be NULL in Finalise()
Also there was observed deadlock if in functions InTransferCallback(), OutTransferCallback() to put sleep(1) in a first string and perform steps above for reproducing the defect. It requires some refactoring in UsbConnection class to avoid such deadlocks.

GetmanetsIrina, 2017-12-22:
Was able to reproduce crash on commit specified in github issue. This time issue happened while OnInTransfer() callback handling

GetmanetsIrina, 2017-12-22:
Tested on commit https://github.com/smartdevicelink/sdl_core/pull/1960 (6d3dca5c43697bd75dc2079ca97bb929cfa1e63c).
Issue is not reproducible.
Transport: USB
Mobile device: Meisu3S
Mobile OS: Android 5.1
Mobile App version: Ford SPT, version:20171120-Nightly-Android
Mobile App type (media/non-media/navi): REMOTE_CONTROL

jacobkeeler, 2018-03-23:
Looks like a duplicate of #1770

jacobkeeler, 2018-08-22:
@GetmanetsIrina I am unable to reproduce this issue, even on the specified commit. Are you still experiencing this on your end? If so, #1771 should be tested as a potential fix to this issue as well.

jacobkeeler, 2019-09-23:
@GetmanetsIrina Closing, feel free to reopen if you are still experiencing this issue

closed 2019-09-23 by jacobkeeler