smartdevicelink/sdl_core #1857: [SDL 0110] Remove QT HMI from SDL Core

theresalech, 2017-11-16:
Proposal: [Remove QT HMI from SDL Core](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0110-remove-qt-hmi-from-sdl-core.md)

> This proposal is for the removal of the QT HMI from the SDL Core repository.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/335

Steering Committee Decision:

> The Steering Committee voted to accept this proposal, thereby agreeing to remove the QT HMI from SDL Core. They agreed that if desired, a proposal can be entered to discuss adding a QT HMI outside of the SDL Core repository, as another HMI.

jacobkeeler, 2018-02-01:
The first task of this proposal has been merged with #1998.

The tasks remaining are:
- Remove QT_HMI_API, as well as any RPCs specific to this API (breaking changes, must be included as part of 5.0.0)
- Remove DBus library and adapter, or refactor it to use the regular HMI API (if necessary).

chinlin0924, 2018-03-05:
hi @jacobkeeler , I saw you fix some dbus compiler error. since the SDL 0110 is accepted, why fix the DBUS issue? or any different decision is made?
thanks

jacobkeeler, 2018-03-20:
@chinlin0924 To my understanding, we are considering the removal of the QT_HMI_API adapter to be a breaking change (due to it containing several RPCs which aren't present in the `HMI_API`, mainly just individual vehicle data RPCs), meaning it needs to be delayed until the next major release. The DBus adapter is the only one that uses this API at this time. 

In addition, after the proposal was accepted, we put out a question to the SDLC whether or not any of the members relied on the presence of the DBus adapter (since it's removal was implicit given the proposal), and we were not given a definitive answer at the time, so we went with a safe alternative of bringing it to at least to a buildable state for the coming release.

I suspect that we will ultimately end up removing this adapter, but as stated before, it will need to be done in the `5.0.0` release.

jacobkeeler, 2018-07-11:
Closing with the merge of #2330

PR #1998 (2018-02-01): Remove QT HMI
Partially Fixes #1857 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
* Removed QT HMI component, along with all references to it.

### Changelog
##### Breaking Changes
* N/A

##### Enhancements
* Removed defunct QT HMI and from this project

### Tasks Remaining:
None

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-07-11 by jacobkeeler