smartdevicelink/sdl_core #1858: [SDL 0108] AutoCompleteList

theresalech, 2017-11-16:
Proposal: [AutoCompleteList](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0108-AutoCompleteList.md)

> The AutoCompleteList will give apps the ability to have multiple listing options while searching for things like destination and information; instead of giving a single suggestion.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/331

theresalech, 2019-07-24:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0108 Revisions - AutoCompleteList](https://github.com/smartdevicelink/sdl_evolution/issues/784).  These changes include updating the feature to support clearing the list by changing the `minsize` for `autoCompleteList` param from 1 to 0.  Please reference issue and markdown file for more details.

jacobkeeler, 2019-07-25:
Closing with the merge of #2959

PR #2959 (2019-07-25): Add autoCompleteList parameter
Fixes #1858 

This PR is **ready** for review.

### Risk
This PR makes **major** API changes.

### Testing Plan
Add Unit Tests and ATF tests

### Summary
Add `autoCompleteList` parameter to KeyboardProperties, replacing `autoCompleteText`

### Changelog

##### Enhancements
* Add `autoCompleteList` parameter to KeyboardProperties

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-07-25 by jacobkeeler