smartdevicelink/sdl_core #1865: SDL does not respond to HMI request with out of bound values

GetmanetsIrina, 2017-11-21:
**Occurrence: Always** 

# Description
SDL does not respond to HMI request with out of bound parameter values/array size.

## Preconditions
- Core, HMI started.
- App is registered on HMI.

## Steps to reproduce
1. HMI sends SDL.GetUserFriendlyMessage request with out of bound value/array size of messageCodes.

## Actual result
SDL does not send response to HMI.

## Expected result
SDL responds with code INVALID_DATA to HMI.

# Environment
SDL commit: 4eb09bf0da82e320723390abd877079af9457d20
OS with SDL: Ubuntu 16.04
ATF version: a6a2a9b2cdec6529cd13dbd11a69c2a7c03cb79c
Transport: TCP

# Attachments
[Defects.zip](https://github.com/smartdevicelink/sdl_core/files/1490570/Defects.zip)
[SDLlog_OutBoundValuesInHMIreq.zip](https://github.com/smartdevicelink/sdl_core/files/1490572/SDLlog_OutBoundValuesInHMIreq.zip)

Note:
Extract attached folder with script to test_scripts folder in ATF build.

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

LuxoftAKutsan, 2017-11-21:
@GetmanetsIrina  please all links to #1869 #1868 #1867 #1866 they probably have same root cause

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to functionality: existence of response from HMI/SDL

HSavynetska, 2018-07-05:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1962
SDL build: develop branch ( ba1da7b )
ATF: develop branch

Status: Defect is reproduced.

ValeriiMalkov, 2018-07-31:
The next PR provided fix this issue: [Pull request](https://github.com/smartdevicelink/sdl_core/pull/2394)

LitvinenkoIra, 2019-12-12:
Defect is reproduced.
SDL branch: develop(e137556)
ATF: develop
Script: smartdevicelink/sdl_atf_test_scripts#1962

LitvinenkoIra, 2019-12-12:
#2394 - rebased to the latest develop
Fix is valid

LitvinenkoIra, 2019-12-13:
Full regression successfully passed
SDL: fix/request_from_hmi_has_para bbd14149
Scripts: develop 6f43bfd
ATF: develop 37da7aba

LitvinenkoIra, 2019-12-13:
@theresalech Please triage this issue

theresalech, 2019-12-16:
@LitvinenkoIra We are currently working on finalizing the roadmap with the Steering Committee, so will keep this issue in consideration for upcoming Core releases.

iCollin, 2020-01-21:
Closed via #2394

closed 2020-01-21 by iCollin