smartdevicelink/sdl_core #1866: SDL does not respond to HMI request with missed mandatory params

GetmanetsIrina, 2017-11-21:
**Occurrence: Always** 

# Description
SDL does not respond to HMI request with missed mandatory params

## Preconditions
- Core, HMI started.
- App is registered on HMI.

## Steps to reproduce
1. HMI sends SDL.GetUserFriendlyMessage request with missed mandatory parameter messageCodes.

## Actual result
SDL does not send response to HMI.

## Expected result
SDL responds with code INVALID_DATA to HMI.

# Environment
SDL commit: 4eb09bf0da82e320723390abd877079af9457d20
OS with SDL: Ubuntu 16.04
ATF version: a6a2a9b2cdec6529cd13dbd11a69c2a7c03cb79c
Transport: TCP

# Attachments
[Defects.zip](https://github.com/smartdevicelink/sdl_core/files/1490790/Defects.zip)
[SDLlog_MandatoryMissingnHMIreq.zip](https://github.com/smartdevicelink/sdl_core/files/1490794/SDLlog_MandatoryMissingnHMIreq.zip)

Note:
Extract attached folder with script to test_scripts folder in ATF build.

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to functionality: existence of response from HMI/SDL

HSavynetska, 2018-07-05:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1961
SDL build: develop branch ( ba1da7b )
ATF: develop branch

Status: Defect is reproduced.

iCollin, 2020-01-21:
Closed via #2394

closed 2020-01-21 by iCollin