smartdevicelink/sdl_core #1871: SDL does not start <DefaultTimeout>+<RPCs_internal_timeout> for all RPCs with own timer

GetmanetsIrina, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not start `DefaultTimeout+RPCs_internal_timeout` for all RPCs with own timer

## Preconditions
- Core, HMI started.
- App is registered on HMI.
- App is activated.

## Steps to reproduce
1.  Mobile app sends RPC with own timeout to SDL.

## Actual result
SDL 
 1. transfers this RPC to HMI  
 2. starts `DefaultTimeout + RPCs_own_timeout` for waiting the response from HMI
 3. transfers `received_resultCode_from_HMI` to mobile app IN CASE HMI responds during `DefaultTimeout + RPCs_own_timeout` timer

## Expected result
SDL does not start `DefaultTimeout + RPCs_own_timeout` for waiting the response from HMI

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: SDL does not provide enough time 
to HMI for processing RPCs with own timeout.

jacobkeeler, 2018-03-08:
Duplicate of #792

closed 2018-03-08 by jacobkeeler