smartdevicelink/sdl_core #1872: PoliciesManager does not allow all requested params in case "parameters" field is omited

GetmanetsIrina, 2017-11-23:
**Occurrence: Always** 

# Description
PoliciesManager does not allow all requested params in case "parameters" field is omited

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.
 - "parameters" field is omited at PolicyTable for used request

## Steps to reproduce
1. Mobile app sends request to SDL and this request is allowed by Policies for this mobile app.

## Actual result
SDL processes RPC as disallowed.

## Expected result
SDL transfers received request with all requested parameters as is to HMI

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to correctness of result code in response from HMI/SDL

AByzhynar, 2018-01-02:
The defect is not reproducible on current sdl_core (develop) branch.
Checked on: https://github.com/smartdevicelink/sdl_core/commit/0b19cf498d1c9972d67b85793dc6b029c26378a0
Branch: https://github.com/smartdevicelink/sdl_core/tree/develop
SDL behaves as expected.

closed 2018-01-02 by AByzhynar