smartdevicelink/sdl_core #1873: PoliciesManager allows all requested params in case "parameters" field is empty

GetmanetsIrina, 2017-11-23:
**Occurrence: Always** 

# Description
PoliciesManager allows all requested params in case "parameters" field is empty

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.
 - "parameters" field is empty at PolicyTable for used request

## Steps to reproduce
1. Mobile app sends request to SDL and this request is allowed by Policies for this mobile app.

## Actual result
SDL transfers received request with all requested parameters as is to HMI.

## Expected result
SDL processes RPC as disallowed.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to policy functionality

GetmanetsIrina, 2017-12-22:
Defect is reproduced on develop branch(0b19cf4).
Fix in #1965,
ATF version for verification https://github.com/smartdevicelink/sdl_atf/commits/develop(ae3c97da2ca77031b645773917b63eff1421c04d),
script for verification https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1873_Parameters_empty_in_policy_table.lua

PR #1969 (2018-02-12): Fix/policies manager allows all requested params in case parameters field is empty
Fixes #1873 

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
Unit tests logic does not need to be updated

### Summary
 - Added check for RPC parameters existence and emptiness in PT
 - Added necessary SDL behaviour for the case when RPC is allowed by policies
   and sent with some parameters. At the same time this RPC has  `parameters` field in PT but it is empty.
- Added different ways to store RPC parameters to policy database. This made to allow differentia [... cut]
closed 2018-02-12 by JackLivio