smartdevicelink/sdl_core #1874: SDL does not respond to mobile app with GENERIC_ERROR in case of invalid HMI response

AByzhynar, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not respond to mobile app with GENERIC_ERROR and correct info string in case of invalid HMI response

## Preconditions
- Core, HMI started.
- App is registered on HMI.
- App is activated

## Steps to reproduce
1. App sends GetVehicleData request with correct parameters.
2. HMI sends invalid response 

Invalid response means:
 - a. params out of bounds
 - b. mandatory params are missing
 - c. params of wrong type

## Actual result
SDL send response to mobile app with 
`{"resultCode":"INVALID_DATA","success":false,"info":"Received invalid data on HMI response"}` .

## Expected result
SDL send response to mobile app with 
`{"resultCode":"GENERIC_ERROR","success":false,"info":"Invalid message received from vehicle"}` .

# Environment
SDL commit: 4eb09bf0da82e320723390abd877079af9457d20
OS with SDL: Ubuntu 16.04
ATF version: a6a2a9b2cdec6529cd13dbd11a69c2a7c03cb79c
Transport: TCP

# Attachments
[GetVehicleData_SpeedParamOutOfUpperBoundInHmiResponse.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1498854/GetVehicleData_SpeedParamOutOfUpperBoundInHmiResponse.tar.gz)
[GetVehicleData_HmiResponseHasNoMandatoryParam.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1498852/GetVehicleData_HmiResponseHasNoMandatoryParam.tar.gz)
[GetVehicleData_SpeedParamHasWrongTypeInHmiResponse.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1498853/GetVehicleData_SpeedParamHasWrongTypeInHmiResponse.tar.gz)

Note:
Extract attached archives with script, atf log and SDL log for every case described above.

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [ ] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to functionality: correctness of result code in response from HMI/SDL

HSavynetska, 2018-07-05:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1960
SDL build: develop branch ( ba1da7b)
ATF: develop branch

Status: Defect is not reproduced.

mked-luxoft, 2019-09-26:
after update in https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1960, this issue is no longer reproducible

closed 2019-09-27 by mked-luxoft