smartdevicelink/sdl_core #1875: SDL does not treat GetVehicleData_response as invalid and does not send GENERIC_ERROR

GetmanetsIrina, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not treat GetVehicleData_response as invalid and does not send GENERIC_ERROR

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. HMI sends GetVehilceData_response with 'gpsData' (and/or 'beltStatus'/ 'deviceStatus'/ 'tireStatus') structure and this structure has at least one parameter with invalid value.

## Actual result
SDL does not send GENERIC_ERROR.

## Expected result
SDL treats GetVehicleData_response as invalid
send GENERIC_ERROR, success:false, info: Invalid response received from vehicle' to mobile app.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to functionality: existence of response from HMI/SDL

HSavynetska, 2019-11-19:
The defect is not reproduced in the current develop

Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1971 
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b  )
ATF: develop branch

closed 2019-11-19 by HSavynetska