smartdevicelink/sdl_core #1876: SDL does not sends GENERIC_ERROR in case of receiving empty 'gpsData' structure

GetmanetsIrina, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not sends GENERIC_ERROR in case of receiving empty 'gpsData' structure

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. HMI sends GetVehilceData_response with 'gpsData' structure and this structure is empty (has no parameters).

## Actual result
SDL does not send GENERIC_ERROR.

## Expected result
SDL must treat GetVehicleData_response as invalid and send 'GENERIC_ERROR, success:false, info: Invalid response received from system' to mobile app, log corresponding error internally.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to functionality: existence of response from HMI/SDL

dboltovskyi, 2018-11-13:
Issue is not reproduced on current develop (https://github.com/smartdevicelink/sdl_core/commit/d36316738785c96dab2ee892762ed08c059fffde)
The reason of that is due to the changes in HMI_API: now `longitudeDegrees` and `latitudeDegrees` parameters of `GPSData` structure are mandatory. 
So if HMI sends response without these parameters SDL treats this response as invalid and replies to mobile with `GENERIC_ERROR` result code after default timeout expires.
Issue is closed.

PR #2268 (2018-08-21): Fix processing of receiving empty 'gpsData' struct
Fixes #1876

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
[x] ATF

### Summary
In case HMI sends empty 'gpsData' structure SDL forwards it to mobile.
But the right flow - SDL should sends GENERIC_ERROR with info 'Invalid response received from system'.
That was implemented

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-11-13 by dboltovskyi