smartdevicelink/sdl_core #1877: SDL does not ignore invalid request/response from hmi and does not log internal error

AByzhynar, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not ignore invalid request/response from hmi and does not log internal error

## Preconditions
- Core, HMI started.
- App is registered on HMI.

## Steps to reproduce
- HMI sends ActivateApp with fake param and invalid `appID` param

## Actual result
N/V

## Expected result
 SDL cuts off fake parameters, log the corresponding error and ignore this response (request)

# Environment
SDL commit: 4eb09bf0da82e320723390abd877079af9457d20
OS with SDL: Ubuntu 16.04
ATF version: a6a2a9b2cdec6529cd13dbd11a69c2a7c03cb79c
Transport: TCP

# Attachments

Note:
Extract attached archives with script, atf log and SDL log for every case described above.

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [ ] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [ ] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to request/responce validation functionality

dboltovskyi, 2018-11-12:
Issue is closed since the same case is described in https://github.com/smartdevicelink/sdl_core/issues/1865

closed 2018-11-12 by dboltovskyi