smartdevicelink/sdl_core #1878: SDL must restore AddCommands in the same order as they were created by mobile app

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL must restore AddCommands in the same order as they were created by mobile app

## Preconditions
- Core, HMI started.
- App is registered on HMI.
- App is activated.

## Steps to reproduce
1.  App successfully added some AddCommands (UI+VR, only UI, only VR)
2.  Ign off – ign on
3.  App registers -> SDL successfully performs HMILevel resumption and data resumption 

## Actual result
The order of resumed AddCommands is different from order being before resumption 

## Expected result
The order of resumed AddCommands must be the same as order being before resumption.
SDL must generate <internal_consecutiveNumber>
assign this <internal_consecutiveNumber> to each AddCommand requested by app,
restore AddCommand by this <internal_consecutiveNumber> during data resumption (meaning: SDL must send UI.AddCommands to HMI by <internal_consecutiveNumber> one by one from resumption list)

# Environment
N/A

# Attachments
N/A

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Low` with reason: The issue is related to order of data restoring during application resumption.

LitvinenkoIra, 2019-12-13:
Defect is reproduced.
SDL branch: develop(e137556)
ATF: develop
Script: smartdevicelink/sdl_atf_test_scripts#2143

LitvinenkoIra, 2019-12-13:
#2515 - rebased to the latest develop
Fix is valid

LitvinenkoIra, 2019-12-13:
Full regression successfully passed
SDL: fix/sdl_must_restore_add_comm 4d4fd660
Scripts: develop 6f43bfd
ATF: develop 37da7aba

LitvinenkoIra, 2019-12-13:
@theresalech Please triage this issue

theresalech, 2019-12-16:
@LitvinenkoIra We are currently working on finalizing the roadmap with the Steering Committee, so will keep this issue in consideration for upcoming Core releases.

iCollin, 2020-01-22:
Closed via #2515

PR #2515 (2020-01-22): SDL restores AddCommands in the order they were created
Fixes #1878 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
SDL generates and assigns internal_id for each AddCommand from mobile app
and restores AddCommands by this internal_id

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-22 by iCollin