smartdevicelink/sdl_core #1879: SDL does not send GetVehilceData_response to mobile app in case of receiving 'gpsData' structure with only one parameter with valid value

AByzhynar, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not send GetVehilceData_response to mobile app in case of receiving 'gpsData' structure
with only one parameter with valid value

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. HMI sends GetVehilceData_response with 'gpsData' structure and this structure has one parameter with valid value.

## Actual result
N/V

## Expected result
SDL must treat GetVehicleData_response as valid and transfer GetVehicleData_response to mobile app

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to critical functionality: existence of response from HMI/SDL.

HSavynetska, 2019-11-20:
The defect is not reproduced on the current develop

Environment:
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop (891497b )
ATF: develop branch
[Script to reproduce the defect_1879.zip](https://github.com/smartdevicelink/sdl_core/files/3869988/Script.to.reproduce.the.defect_1879.zip)

closed 2019-11-20 by HSavynetska