smartdevicelink/sdl_core #1880: SDL does not send GENERIC_ERROR in case HMI does NOT respond during <DefaultTimeout>+<RPCs_internal_timeout>

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL must send GENERIC_ERROR to app in case HMI does NOT respond during <DefaultTimeout>+<RPCs_internal_timeout> for all RPCs with own timer

## Preconditions
- Core, HMI started.
- App is registered on HMI.
- App is activated.

## Steps to reproduce
1.  SDL transfers *RPC with own timeout from mobile app to HMI 
and HMI does NOT respond during <DefaultTimeout> + <*RPCs_own_timeout>

## Actual result
N/A

## Expected result
SDL must respond 'GENERIC_ERROR, success:false' to mobile app

# Environment
N/A

# Attachments
N/A

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to critical functionality: existence of response from HMI/SDL.

JackLivio, 2018-03-06:
Similar issue to #1871 and #792

jacobkeeler, 2018-03-08:
Looks like a duplicate of #792, but with the specification of `2*RPCs_internal_timeout` added to the default timeout instead. Why is this?

yang1070, 2018-05-14:
@LitvinenkoIra I have the same question as Jacob here, Why is 2*RPCs_internal_timeout + default?

GetmanetsIrina, 2018-05-15:
@JackLivio , @yang1070 , according to requirements:
SDL must start `DefaultTimeout(from .ini file) + RPCs_internal_timeout` for waiting the response from HMI side.
_RPCs_internal_timeout for RPCs:_
a. Alert (“duration” param)
b. PerformInteraction (“timeout” param -> increased two times for BOTH and MANUAL_ONLY modes)
c. Slider (“timeout” param)
d. ScrollableMessage (“timeout” param)

So described case `DefaultTimeout+2*RPCs_internal_timeout` is related to  `PerformInteraction` RPC with `BOTH` and `MANUAL_ONLY` modes

yang1070, 2018-05-16:
@GetmanetsIrina 
I'm kind of understand the requirement in the case of `PerformInteraction` RPC with` BOTH` mode, as the work flow is to show VR portion, after some system defined time out  (handled by the platform), it shows the MAUNAL portion (the timeout paramter in RPC). So the sdl wait time shall be longer than DefaultTimeout+RPCs_internal_timeout due to the added VR portion. But it does not specify how much longer as we don't known what's the timeout for VR portion. This might be the reason that we treat it as the same as the timeout parameter. Therefore the result DefaultTimeout+2*RPCs_internal_timeout.

But why`MANUAL_ONLY` mode also need DefaultTimeout+2*RPCs_internal_timeout?

From the mobile API, I get the following description in `PerformInteraction` RPC
```xml
        <param name="timeout" type="Integer" minvalue="5000" maxvalue="100000" defvalue="10000" mandatory="false">
            <description>
                Timeout in milliseconds.
                If omitted a standard value of 10000 milliseconds is used.
                Applies only to the menu portion of the interaction. The VR timeout will be handled by the platform.
            </description>
        </param>
```

and from integration guideline, I get the following description about timeout
```
Timeout in milliseconds. 
The amount of time the HMI must wait for the   User to make a choice. 
After the time is up HMI respond with timed_out result   code.
```

GetmanetsIrina, 2018-05-29:
@yang1070 , @jacobkeeler , @JackLivio 
According to discussion in emails the expected behavior is the same, updated with more details:

### Steps to reproduce
1. SDL transfers RPC with own timeout from mobile app to HMI _with the same timeout value received from mobile app_
    and HMI does NOT respond during DefaultTimeout+*RPCs_own_timeout

### Note:
RPCs_own_timeout for RPCs:
a. Alert (“duration” param)
b. PerformInteraction (“timeout” param -> increased two times for BOTH and MANUAL_ONLY modes)
c. Slider (“timeout” param)
d. ScrollableMessage (“timeout” param)

JackLivio, 2019-10-16:
Fixed by #2495

PR #2495 (2019-10-16): Fix/sdl must send generic error hmi doesnt respond during timeout
Fixes #1880 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
This PR provided bugfix when SDL doesn't respond during `DefaultTimeout`+`RPCs_internal_timeout` for all RPCs with own timer.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2019-10-16 by JackLivio