smartdevicelink/sdl_core #1881: SDL does not send notification to app right after this app changes level from NONE to any other

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL must send notification to app right after this app changes level from NONE to any other

## Preconditions
- Core, HMI started.

## Steps to reproduce
1. mobile app registers and gets NONE HMILevel and SDL receives OnDriverDistraction (<state>) fom HMI

## Actual result
N/A

## Expected result
SDL must transfer the last known (actual) OnDriverDistraction (<state>) to this mobile app right after this mobile app changes HMILevel to any other than NONE.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: Currently the newly registered applications know nothing about Driver Distraction mode, but only after it changes, the issue fixes this case.

dboltovskyi, 2017-12-22:
Issue is reproduced on develop (0b19cf4) and fixed in https://github.com/smartdevicelink/sdl_core/pull/1970

PR #1970 (2018-02-12): Implement new behavior for OnDriverDistraction 
Fixes #1881

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
Updated appropriate unit tests

### Summary
According to the new behavior SDL has to postpone OnDriverDistraction
for the application which are in NONE HMI level and transfer the last known (actual) OnDriverDistraction (<state>) to this mobile app right after this mobile app changes HMILevel to any other than NONE.
Note: Whether the app receives the notification in current HMILevel is defined by app's assigned Policies. Often, OnDriverD [... cut]
closed 2018-02-12 by JackLivio