smartdevicelink/sdl_core #1883: ChoiceSet being processed in DeleteInteractionChoceSet

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
ChoiceSet being processed in DeleteInteractionChoceSet

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1.  HMI processes VR.DeleteCommands_request related to choiceSetID_1 AND app sends PerformInteractionChoiceSet with choiceSetID_1 

## Actual result
N/A

## Expected result
 SDL must respond REJECTED (success:false) to this PerfromInteraction

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to Perform interaction RPC functionality. Supposed that SDL responds with wrong result code.

dboltovskyi, 2021-09-23:
Issue is still applicable for https://github.com/smartdevicelink/sdl_core/tree/release/8.0.0 (de58a41)

theresalech, 2021-11-09:
@dboltovskyi, can you please advise when merge conflicts for PR #2613 have been resolved and the PR is ready for Livio's review?

dboltovskyi, 2021-11-09:
@theresalech Please notice initial fix for this issue in [#2613](https://github.com/smartdevicelink/sdl_core/pull/2613) is pretty old. We're updating and re-testing it now. Our plan is to re-submit new version in a couple of days.

iCollin, 2022-01-05:
Hello @dboltovskyi, could you please help me understand what this change would be used for? What would Mobile do differently if it receives a REJECTED responses versus the current INVALID_ID? From a quick look, I don't see the value of differentiating these pending delete choice sets.

dboltovskyi, 2022-01-06:
@VjKlepikov Could you please support with the question from @iCollin. Thank you.

VjKlepikov, 2022-01-06:
@iCollin According to requirements [APPLINK-14954]( https://adc.luxoft.com/jira/browse/APPLINK-14954)

> In case HMI processes VR.DeleteCommands_request related to choiceSetID_1 AND app sends PerformInteractionChoiceSet with choiceSetID_1 -> SDL must respond REJECTED (success:false) to this PerfromInteraction

Also, I believe the mobile does not send another PerformInteractionChoiceSet with the same ID in case it receives "INVALID_ID" 
If the "REJECTED" would be the response in which tells the mobile app it should maybe try the request again with the same ID because DeleteInteractionChoceSet could be rejected

iCollin, 2022-01-06:
@VjKlepikov thank you for your explanation.

I understand there is a requirement but I am not sure it is still applicable considering Core has unconditionally replied success for DeleteInteractionChoiceSet until recently. It looks like the libraries [do not currently consider the response code to retry a PerformInteraction](https://github.com/smartdevicelink/sdl_javascript_suite/blob/master/lib/js/src/manager/screen/choiceset/_PreloadPresentChoicesOperation.js#L679).

I agree it is possible mobile could retry if it gets REJECTED response, but when would the second PerformInteraction request succeed? The SDL HMI [sends success unless the choice set is in use](https://github.com/smartdevicelink/sdl_hmi/blob/master/app/model/sdl/Abstract/Model.js#L1783), but an in use check already exists in Core. I can't really imagine a use case where a second PerformInteraction request would succeed.

I am still struggling to understand the value of tracking if a choice set is pending delete. Do my concerns make sense to you?

VjKlepikov, 2022-01-17:
@iCollin 
I believe that the second PerformInteraction request would succeed in case HMI does not respond to VR.DeleteCommand triggered by DeleteInteractionChoiceSet during first request. I agree that the libraries do not currently consider the response code to retry a PerformInteraction but it can be changed by mobile developers
[image]

iCollin, 2022-01-25:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3826

PR #3826 (2022-01-25): Fix PerformInteractionChoiceSet request behavior
Fixes #1883

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by ATF test scripts, the new unit test is also added

### Summary
There is a case when DeleteInteractionChoiceSet request is sent to HMI, and SDL is waiting for response. If during this waiting period PerformInteractionChoiceSet request is sent, according to requirements SDL should reject such request.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/v [... cut]
closed 2022-01-25 by iCollin