smartdevicelink/sdl_core #1884: SDL behavior in case HMI responds with WARNINGS to at least one VR.DeleteCommands AND app sends PerfromInteraction related to this ChoiceSet

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL behavior in case HMI responds with WARNINGS to at least one VR.DeleteCommands AND app sends PerfromInteraction related to this ChoiceSet

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. In case of processing DeleteInteractionChoiceSet HMI responds with WARNINGS to at least one VR.DeleteCommands_request related to ChoiceSetID_1 and app sends PerformInteraction for removed choiceSetID 

## Actual result
N/A

## Expected result
SDL must respond PerformInteraction (REJECTED, success:false) to mobile app

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to Perform interaction RPC functionality. Supposed that SDL responds with wrong result code.

iCollin, 2020-05-08:
Hello @LitvinenkoIra , does https://github.com/smartdevicelink/sdl_core/pull/2772 fix this issue?

iCollin, 2020-05-13:
@LitvinenkoIra I do not see any way to get a WARNINGS response to a `VR.DeleteCommand`. Can you please provide more reproduction steps?

iCollin, 2020-05-20:
I am closing this issue as not reproducible. If you are still seeing this issue please let me know.

closed 2020-05-20 by iCollin