smartdevicelink/sdl_core #1885: SDL does not cut off unknown RPC and/or param from PTU

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL must cut off unknown RPC and/or param from PTU

## Preconditions
- Core, HMI started.
- App is registered on HMI.
- App is activated.

## Steps to reproduce
1.  In case SDL received UpdatedPT with at least one <unknown_parameter> or <unknown_RPC>

## Actual result
N/A

## Expected result
SDL must cut off <unknown_parameter> or <unknown_RPC>
continue validating received PTU without <unknown_parameter> or <unknown_RPC>
merge valid Updated PT without <unknown_parameter> or <unknown_RPC> with LocalPT in case of no other failures

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Low` with reason: The issue is related to PTU functionality and assumed that currently SDL just ignores unknown RPC and parameters.

LuxoftAKutsan, 2017-12-27:
Fix available https://github.com/smartdevicelink/sdl_core/pull/1972

LuxoftAKutsan, 2017-12-27:
Issue is related to #1921

AStasiuk, 2018-07-31:
Closed as duplicate of #1921

closed 2018-07-31 by AStasiuk