smartdevicelink/sdl_core #1887: [OnVehicleData] PoliciesManager allows all requested params in case "parameters" field is empty

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
PoliciesManager allows all requested params in case "parameters" field is empty

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. In case SDL receives OnVehicleData notification from HMI
and this notification is allowed by Policies for this mobile app
and "parameters" field is **empty** at PolicyTable for OnVehicleData notification

## Actual result
N/A

## Expected result
SDL must log corresponding error internally
SDL must NOT transfer this notification to mobile app

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Medium` with reason: The issue is related to notification about vehicle information and provides not allowed information.

Ypostolov, 2018-09-12:
SDL build : develop branch (28235f88a6733106a7e8a0d490dfb16a5f374a33)
ATF: develop branch

Status: Defect is not reproduced.

mked-luxoft, 2019-09-27:
SDL : e356886876 
ATF  : 05ba940

not reproducible

closed 2019-09-27 by mked-luxoft