smartdevicelink/sdl_core #1888: SDL behavior in case no "certificate" in "module_config" section of PT

AStasiuk, 2017-11-23:
**Occurrence: Always** 

# Description
SDL behavior in case no "certificate" in "module_config" section of Policy Table

## Preconditions
 - SDL and HMI are started.
 - App is registered.

## Steps to reproduce
1. In case SDL starts PolicyTableUpdate in case of no "certificate" at "module_config" section at LocalPT and PolicyTableUpdate is valid and brings "certificate" at "module_config" section and app sends StartService (<any_serviceType>, encypted=true) to SDL

## Actual result
N/A

## Expected result
SDL must start TLSHandshake sequence (by sending TLS Hello to this mobile app

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to secure service start logic.

dboltovskyi, 2017-12-27:
Issue is reproduced on develop branch (0b19cf4)

jacobkeeler, 2018-01-31:
Verified that this is fixed by #1971.

jacobkeeler, 2018-01-31:
Fixed with the merge of #1971

closed 2018-01-31 by jacobkeeler