smartdevicelink/sdl_core #1889: SDL does not ignore OnVehilceData_notification in case of receiving empty 'gpsData' structure

AByzhynar, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not ignore OnVehilceData_notification in case of receiving empty 'gpsData' structure
without any parameters. 

## Preconditions
1. SDL and HMI are started.
2. App is registered and activated.
3. App is subscribed for some vehicle data

## Steps to reproduce
1. HMI sends OnVehilceData_notification with empty 'gpsData' structure

## Actual result
N/V

## Expected result
SDL must treat OnVehicleData_notification as invalid, log internal error and does not transfer OnVehicleData_notification to mobile app

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `Low` with reason: The issue is related to notification without incorrect data.

HSavynetska, 2019-11-20:
The defect is not reproduced on the current develop

Environment:
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop (891497b )
ATF: develop branch
[Script to reproduce the defect.zip](https://github.com/smartdevicelink/sdl_core/files/3868759/Script.to.reproduce.the.defect.zip)

closed 2019-11-20 by HSavynetska