smartdevicelink/sdl_core #1890: SDL does not respond GENERIC_ERROR in case no response on VR.PerformInteraction during timeout from HMI

AByzhynar, 2017-11-23:
**Occurrence: Always** 

# Description
SDL does not respond **GENERIC_ERROR** in case no response on **VR.PerformInteraction** during timeout from HMI

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. App sends PerformInteraction(params)
2. SDL sends UI.PerformInteraction and VR.PerformInteraction to HMI
3. SDL starts default timeouts for UI and VR portions (defined in .ini file)
4. HMI responds UI.PerformInteraction(SUCCESS) to SDL
5. HMI doesn't send VR.PerformInteraction response or OnResetTimeout notification during timeout

## Actual result
N/V

## Expected result
SDL sends PerformInteraction (GENERIC_ERROR) to mobile app

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set High with reason: The issue is related to one of common used RPCs.

HSavynetska, 2018-06-26:
Current issue is duplicate of  #1880 and can be closed.

closed 2018-11-14 by LitvinenkoIra