smartdevicelink/sdl_core #1892: SDL must start heartbeat only after first Heartbeat request from mobile app

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL must start heartbeat only after first Heartbeat request from mobile app

## Preconditions
 - SDL and HMI are started.
-  mobile app successfully connects to SDL over protocol v3 or higher
-  the value of "HeartBeat" param at .ini file is more than zero

## Steps to reproduce
1. App sends first HeartBeat request by itself over control service to SDL

## Actual result
SDL start HeartBeat process right after first StartService_request from mobile app

## Expected result
SDL must respond HeartBeat_ACK over control service to mobile app start HeartBeat timeout (defined at .ini file)
SDL must NOT start HeartBeat process right after first StartService_request from mobile app (as currently implemented)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to connection/session logic
Heartbeat should be applied only after establishing service

GetmanetsIrina, 2017-12-19:
Fix in https://github.com/AKalinich-Luxoft/sdl_core/pull/2 (5329f5e70ce3b93dcaee1bc09111b17a3f13b015),
 ATF version for verification https://github.com/smartdevicelink/sdl_atf/pull/110 (58e36f1c18f6575926db58b0bf60cf99d11cbf2b) with HB fix,
script for verification  https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1892_HB_from_SDL_after_receiving_HB_from_mobile.lua.

PR #3572 (2020-11-13): Fix session heartbeat starting
Updates copy of pr #1968

Fixes #1892

### Summary
Updated pr with resolved merge conflicts. SDL will not automatically send a heartbeat to an app after a registration. App must send first heartbeat otherwise heartbeat is assumed to be unsupported.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #1968 (2020-11-13): Fix session heartbeat starting
Fixes #1892

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Changes in PR are already covered within existing unit tests.

### Summary
In the current implementation SDL starts session heartbeat right after application registration.
However the correct behavior is to start session heartbeat after first heartbeat request from mobile app.

### Changelog
* Removed starting session heartbeat on app registration
* Added session heartbeat start on frist heartbeat request from mobile app
* Added check to prevent starting of [... cut]
closed 2020-11-13 by JackLivio