smartdevicelink/sdl_core #1893: SDL must close only session in case mobile app does not answer on Heartbeat_request

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL must close only session in case mobile app does not answer on Heartbeat_request

## Preconditions
 - SDL and HMI are started.

## Steps to reproduce
1. In case SDL has one connection with two sessions (= app_1 and app_2)
and app_1 does not send anything (RPC, data streaming) and does not respond on Heartbeat request from SDL

## Actual result
N/A

## Expected result
SDL must close session ONLY with this app_1 (e.g. connection with active app_2 (active session) must continue working)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

aderiabin, 2017-12-01:
Contributor priority is set `High` with reason: The issue is related to connection/session logic
Mobile applications should work independently one from other.

LitvinenkoIra, 2017-12-04:
Works correctly on current develop (8c94933)
Script for verifying and testing report [1893.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1527442/1893.tar.gz)

LuxoftAKutsan, 2017-12-11:
@jacobkeeler @JackLivio please close the issue.

LuxoftAKutsan, 2017-12-14:
HeartBeatMonitor is in charge of monitoring heartbeat timeout. There is separate  HeartBeatMonitor for each connection.
It has separate own thread that executing `HeartBeatMonitor::Process` function. 
HeartBeatMonitor handles session numbers and monitors heartbeat for each session : 
```
class HeartBeatMonitor : public threads::ThreadDelegate {
...
   * \brief add and remove session
   */
  void AddSession(uint8_t session_id);
  void RemoveSession(uint8_t session_id);
...
```
```
void HeartBeatMonitor::Process() {
  sessions_list_lock_.Acquire();
  SessionMap::iterator it = sessions_.begin();
  while (it != sessions_.end()) {
  ...
```
And in case if Hearbeat timeout elapsed HeartBeatMonitor will close certain session  : 
```cpp
void HeartBeatMonitor::Process() {
...
    if (state.HasTimeoutElapsed()) {
      const uint8_t session_id = it->first;
      if (state.IsReadyToClose()) {
        LOG4CXX_WARN(logger_, "Will close session");
        sessions_list_lock_.Release();
        RemoveSession(session_id);
        connection_->CloseSession(session_id);
        sessions_list_lock_.Acquire();
...
}
```

dboltovskyi, 2017-12-20:
There is some extension for the initial requirement:
SDL should not close mobile connection even if last session on this connection had been closed.

Currently on a head of develop SDL close such connection, so issue is reproduced.

Description of the issue is updated.
Test script is also updated accordingly.

mrapitis, 2017-12-21:
Requires additional testing with the updated script that is provided.

GetmanetsIrina, 2017-12-22:
Defect is reproduced on develop branch(0b19cf498d1c9972d67b85793dc6b029c26378a0).
Fix in https://github.com/smartdevicelink/sdl_core/pull/1965(0abb3c77bd9fad9a8070e4a9b041536c4356af5f),
ATF version for verification https://github.com/smartdevicelink/sdl_atf/commits/develop(ae3c97da2ca77031b645773917b63eff1421c04d) with HB fix,
script for verification https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1893_ATF_HeartBeat_App_does_not_send_HB_and_does_not_respond.lua.

PR #1965 (2018-09-04): Fix/wrong connection closure

Fixes #1893 

This PR is **[ready]** for review.

### Risk
This PR makes **[no]** API changes.

### Testing Plan
Appropriate unit tests added/updated

### Summary
 - Removed connection closure in case of last session in the connection was closed.
   All connections should be removed when device would be unplugged.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-09-04 by JackLivio